microsoft/AirSim #1345: Urban environment in linux

ndeshp2s, 2018-08-23:
Hello all,

I am new to AirSim and wanted to know if some one has already used a basic urban environment for autonomous vehicle testing in AirSim for LINUX? I could find Neighbourhood and CityEnvironment binaries for windows. But nothing for Linux. 

What could be the best possible option to simulate an urban environment in AirSim for Linux.

thank you.

sytelus, 2018-10-17:
We have released City Environment for Linux. More environments coming soon.

closed 2018-10-17 by sytelus

petergerten, 2018-11-02:
@sytelus Coudl you tell us where to find the City Environment for Linux?

madratman, 2018-11-02:
@petergerten Go to releases -> v1.2.0-Linux. 
Alternatively, here's the link: https://github.com/Microsoft/AirSim/releases/tag/untagged-a395e93dd037d937537a 
You should see two zip files : `CityEnviron.zip.001` and `CityEnviron.zip.002`. There are details in the subtext of the release.

petergerten, 2018-11-03:
@madratman the link is not working for me (404 error) and I can't find " v1.2.0-Linux." when I manually browse releases.

ndeshp2s, 2018-11-13:
Thanks @sytelus for the update. However, the link does not work for me either @madratman.

nightduck, 2020-08-26:
Picking this up. The tag is [here](https://github.com/microsoft/AirSim/releases/tag/v1.2.0Linux). I was able to unzip them with 7zip, but upon running CityEnviron.sh, I get: error while loading shared libraries: libopenvr_api.so: cannot open shared object file: No such file or directory.

Some of the other environments have an "Engine" folder with them which contains this file, but borrowing their copy doesn't fix the issue.