microsoft/AirSim #1346: Does the native recording feature support recording from all vehicles in a scene?

saihv, 2018-08-24:
The code in RecordingThread.cpp, does it handle multiple drones/cars?

saihv, 2018-08-24:
Answer: it doesn't. But it isn't hard to extend it to operate with multiple vehicles, by passing vectors of ImageCapture and Kinematics objects to the record thread (instead of a single object from vehicle 0).

closed 2018-08-24 by saihv

arincbulgur, 2019-08-01:
@saihv 

I am using [kshitiz38](https://github.com/kshitiz38/AirSim)'s version to work with both a drone and ground vehicle. However, I also want to record the kinematics of multiple vehicles. I tried to use your modified RecordingThread.cpp in the [link](https://github.com/saihv/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Recording/RecordingThread.cpp) to build the AirSim. It created the plugin bits without an error. However, when I tried to add the plugins into my unrealproject, it's unable to build the project.

Is your modified RecordingThread.cpp not compatible with the current APIs? Or should I also change some other codes? I am a beginner level of C++ user. Can you help me about this issue?