microsoft/AirSim #1348: Some problem about the AirSimNH

slyviacassell, 2018-08-24:
I download the released assets AirSimNH and extract it on my computer. But when I run the **run.bat** but it shows: 
WARNING:tornado.general:Connect error on fd 264: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 264: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 264: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 264: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 264: WSAECONNREFUSED
Waiting for connection: Traceback (most recent call last):
  File ".\image_collection.py", line 30, in <module>
    client.confirmConnection()
  File "C:\Users\Asus\Desktop\AirSimTensorFlow\AirSimClient.py", line 160, in confirmConnection
    home = self.getHomeGeoPoint()
  File "C:\Users\Asus\Desktop\AirSimTensorFlow\AirSimClient.py", line 169, in getHomeGeoPoint
    return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
  File "F:\Python3.5.4\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "F:\Python3.5.4\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit
I know the AirSim server does not work, but does anyone can help me to fix it?

slyviacassell, 2018-08-24:
[image]

madratman, 2018-12-18:
Hey @slyviacassell this happens if you run the python script before the unreal binary. Please let us know if were/are you able to get it working

closed 2018-12-18 by madratman

ArthurYdalgo, 2021-07-10:
I replied to a [similar issue](https://github.com/microsoft/AirSim/issues/1949) [what could be the solution](https://github.com/microsoft/AirSim/issues/1949#issuecomment-877612274)