microsoft/AirSim #1351: Cannot Change image resolution

hxlee309, 2018-08-27:
Hi,

I want to set the image resolution to be 640 x 480. The problem is that I am still getting 256 x 148 resolution, which is very annoying. I have already changed the settings.json file in the following folder: C:\Users\hxlee\Documents\AirSim. And here's the file:
{
  "SettingsVersion": 1.2,
  "CaptureSettings": [
      { 
      	"ImageType": 0,
      	"FOV_Degrees": 120,
        "Width": 50, 
        "Height": 50
      },
      { 
      	"ImageType": 1,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 2,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 3,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 4,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 5,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 6,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      },
      { 
      	"ImageType": 7,
      	"FOV_Degrees": 120,
        "Width": 640, 
        "Height": 480
      }
],
  "SimMode":"Car"
}

Here's my python script for saving images:
import setup_path 
import airsim

import time
import os
import numpy as np
import io
from PIL import Image

# connect to the AirSim simulator 
client = airsim.CarClient()
client.confirmConnection()
print("reset")
client.reset()
client.enableApiControl(True)
car_controls = airsim.CarControls()

for idx in range(1):
    # get state of the car
    car_state = client.getCarState()
    print("Speed %d, Gear %d" % (car_state.speed, car_state.gear))

    count = 0
    filename = '/mnt/e/AirSim/AirSim-1.2.0/PythonClient/image_temp/'
    # go forward
    while count < 100:
        car_controls.throttle = 0.5
        car_controls.steering = 0
        client.setCarControls(car_controls)
        print("Go Forward")
        png_image = client.simGetImage("0", airsim.ImageType.Scene)
        print("png_image type is: ", type(png_image))
        print("png_image shape is: ", len(png_image))
        im = Image.open(io.BytesIO(png_image))
        im.save(filename + 'png' + str(count) + ".png")
        count += 1
        
    # go reverse
    car_controls.throttle = -0.5
    car_controls.is_manual_gear = True;
    car_controls.manual_gear = -1
    car_controls.steering = 0
    client.setCarControls(car_controls)
    print("Go reverse, steer right")
    time.sleep(3)   # let car drive a bit
    car_controls.is_manual_gear = False; # change back gear to auto
    car_controls.manual_gear = 0  

    # apply breaks
    car_controls.brake = 1
    client.setCarControls(car_controls)
    print("Apply break")
    time.sleep(3)   # let car drive a bit
    car_controls.brake = 0 #remove break

#restore to original state
client.reset()

client.enableApiControl(False)

Any help will be appreciated.

Thanks,

Hanxiang

sytelus, 2018-10-17:
With AirSim 1.2 the settings schema is changed. See [CameraDefaults settings](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md).

madratman, 2018-12-07:
@hxlee309 too late, but you can do:

```
{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,

"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 0,
"Width": 640,
"Height": 480,
"FOV_Degrees": 120
}
]
}
}
```

closed 2018-12-07 by madratman

vishal-prgmr, 2019-04-23:
@sytelus , My .json is as follows:
I am running the latest master branch with UE 4.18 on Win7.
Have the same issue that the resolution is not getting changed. 
`{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,

    "SimMode": "Car",
	"RecordUIVisible": true,
	"LogMessagesVisible": true,
	"Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "front_center", "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
    ]
	},
	"CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       },
	   {
        "ImageType": 1,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       },
	   {
        "ImageType": 5,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       }
	   
    ]
  },
	"SubWindows": [
					{"WindowID": 0, "ImageType": 0, "CameraName": "front_center", "Visible": true},
					{"WindowID": 1, "ImageType": 1, "CameraName": "front_center", "Visible": true},
					{"WindowID": 2, "ImageType": 5, "CameraName": "front_center", "Visible": true}
				  ],
     "Vehicles": {
        "EgoCar": {
            "VehicleType": "PhysXCar",
            "AutoCreate": true,
			"Cameras": {
				"front_center": {
				"CaptureSettings": [
				  {
					"ImageType": 0,
					"width": 1280,
					"Height": 960,
					"FOV_Degrees": 52.75
					
				  }
				],
				"NoiseSettings": [
				  {
					"Enabled": false,
					"ImageType": 0,

					"RandContrib": 0.2,
					"RandSpeed": 100000.0,
					"RandSize": 500.0,
					"RandDensity": 2,

					"HorzWaveContrib":0.03,         
					"HorzWaveStrength": 0.08,
					"HorzWaveVertSize": 1.0,
					"HorzWaveScreenSize": 1.0,

					"HorzNoiseLinesContrib": 1.0,
					"HorzNoiseLinesDensityY": 0.01,
					"HorzNoiseLinesDensityXY": 0.5,

					"HorzDistortionContrib": 1.0,
					"HorzDistortionStrength": 0.002
				  }
				],
		
				
				"X": 0, "Y": 0, "Z": -2,
				"Pitch": 0, "Roll": 0, "Yaw": 0    
				}
			  },
			
			"Sensors": {
                "LidarSensor1": { 
                    "SensorType": 6,
                    "Enabled" : true,
                    "NumberOfChannels": 16,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 100000,
                    "X": 0, "Y": 0, "Z": -2.5,
                    "Roll": 0, "Pitch": 0, "Yaw" : 0,
                    "VerticalFOVUpper": -15,
                    "VerticalFOVLower": -25,
                    "HorizontalFOVStart": -360,
                    "HorizontalFOVEnd": 360,
                    "DrawDebugPoints": true,
                    "DataFrame": "SensorLocalFrame"
                }		
            }
        }
    }
}`

vishal-prgmr, 2019-04-23:
@sytelus ,
I am running the latest master branch with UE 4.18 on Win7.
Have the same issue that the resolution is not getting changed. 
 My .json is as follows:

`{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,

    "SimMode": "Car",
	"RecordUIVisible": true,
	"LogMessagesVisible": true,
	"Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "front_center", "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
    ]
	},
	"CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       },
	   {
        "ImageType": 1,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       },
	   {
        "ImageType": 5,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 52.75
       }
	   
    ]
  }
`

cnaveenf, 2019-06-10:
Is this issue resolved? I  am interested in a solution to this problem.
It would be nice to be able to change the paramters of the camera model in unreal through Airsim apis