microsoft/AirSim #1353: Errors on building AirSim v1.2.0 and problems with simulation and PX4 SITL

danijc8196, 2018-08-27:
Hi @lovettchris @sytelus @ashkapoor and the whole community!

I am a Telecommunication Engineering student from Spain working on my final thesis, which is based on AirSim. The experiment basically consists in setting up AirSim with the PX4 autopilot and ROS, and train a multirotor with some kind of Reinforcement Learning algorithm. The objective is getting a simulation system (based on AirSim) which allows migrate all the code directly from the simulation to the real multirotor (using PX4 too) without changing anything (or at most, a few details).

Recently I wanted to set up the latest version of AirSim (v1.2.0), but I find 2 kind of problems which are blocking me at all. I searched a lot on the web and between the AirSim issues, I removed and repeated the installation several times, and tried to find the problems by myself, but at the end I am so lost and need a lot of help. Here is my explanation:

### **First of all**
I am working on Ubuntu 16.04 (I need linux for setting up ROS) and PX4 SITL.
I set up the Unreal Engine 4.18.3 and forked the AirSim repo before checking out the software version 1.2.0. Then I updated cmake to 3.10.2 version, and finally, I followed the instructions on https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md for setting up and building AirSim.
At this point all seemed to be OK!

### **First problem**
But when I wanted to set up the Blocks environment some problems appeared. I opened the Blocks.uproject and selected "More options" and "Convert-in-place" as it sais on the build_linux file.
After a while working, the Unreal raises this error:
[image]

Furthermore, there was some compilation errors on the shell. I copied them to the following file (i feel easier to read them in a separate file than doing it from the issue's body)
[errors_from_shell.txt](https://github.com/Microsoft/AirSim/files/2325267/errors_from_shell.txt)

After some time examining these errors I tried to solve them by myself. It was easier than expected, and just with these two changes, the compilation errors desappeared:

In file:

> MyAirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:602:16:

Changed `static std::unique_ptr<VehicleSetting> createPX4VehicleSetting(const Settings& settings_json)` by `static std::unique_ptr<PX4VehicleSetting> createPX4VehicleSetting(const Settings& settings_json)`

And then, in file:

> MyAirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/VehicleApiBase.hpp:87:29

Added a constructor `RCData(){};` after the arguments of the struct.

At this point, all was working OK (or seemed it) and the Unreal ran the Blocks environment.

### **Second part: second problem**
I launched the PX4 and ran the simulation. No errors appeared. But when I wrote `px4> commander takeoff` the multirotor started running and went up climbing without limit.
After analyzing mavlink messages and simulation behavior, I wanted to take this screenshot:
[image]

In the capture above the multirotor is climbing all the time, while the relative height measured is between 1.9 and 2 meters. I read the height from MavLink analyzer, ROS topics and QGroundControl (noted in cyan).
 My understanding is that measures (they came from the simulation through mavlink, if i am not wrong) are OK, as they should be. But the multirotor simulation is wrong, like understanding that the 2 meters height is never reached, so it keeps climbing. It is so strange for me.

I tried to give you all the information that I could find about my problems. I hope anyone can understand the problem and help me and future related issues.

Finally, I would want to congratulate all the Microsoft team members working on this huge project. This is something brilliant!
Thanks in advance, and kind regards.

sytelus, 2018-09-17:
I think these errors have already been fixed in latest version. Could you please try again building it from latest code and let us know if it is still happening?

lovettchris, 2018-12-18:
The issue of the drone climbing infinitely is fixed.

closed 2018-12-18 by madratman