microsoft/AirSim #1360: Data Logging Code

SoumiDas, 2018-08-29:
Hi,

I'm new to AirSim and I wish to obtain some information. I am using City Environment and want to manually drive the car and record the images in different ImageTypes viz. Scene , Segmentation. Which is the file I should be playing around with for manually driving i.e after recording and obtaining the different types of images.

Thanks!

thias15, 2018-09-01:
Set up the settings file with the cameras you wish to record according to docs. Then hit R to start/stop recording. (Or the big red button)

SoumiDas, 2018-09-01:
Thanks a lot.

So can we record the steering, throttle and speed values too via the
settings.json file? I checked with the hello_car.py code and we can fetch
those control values but is it possible through only the settings file?

sytelus, 2018-09-17:
Take a look at this: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#recording

madratman, 2018-12-18:
@SoumiDas did that answer your question?

closed 2018-12-18 by madratman