microsoft/AirSim #1367: Swapping out segmentation color pallet breaks segmentation

benjamin-lange, 2018-09-03:
I use a customized segmentation color pallet. It worked fine. Couple of days ago I updated Airsim. After the update I tried restoring my custom segmentation using my pallet.
The segmentation view with the original pallet:
[image]

The segmentation view with my pallet:
[image]

My segmentation file:
[image]

Is anybody experiencing the same problem with the current Airsim version?
I have not changed my customized pallet in any way I simply copy pasted it after installing the new Airsim version.

benjamin-lange, 2018-09-03:
I filled the custom pallet with more colors and suddenly the segmentation view gained more colors.
But the colors do not correspond to the "CustomDepth Stencil Values" I defined.
It seems there is some kind of offset or randomness when picking the segmentation colors.
I will check the stencil IDs next.

vaschindler, 2018-09-13:
How did you manage to use your own color palett? I also need to change the colors to fit my task. Could you give me a hint?

benjamin-lange, 2018-09-14:
First of all I am using a custom environment with my own scene and models.
If you do too, then the following could work out for you:

You need to swap out the "seg_color_pallet" in the "Unreal/Plugins/AirSim/Content/HUDAssets/".
Then you need to reimport the seg_color_pallet texture in the UE4 editor (AirSim/Content/HUDAssets):
[image]
[image]

After this you need to verify that the texture is updated in the SegmentationMaterial (AirSim/Content/HUDAssets):
[image]

Now you need to make sure that Custom Depth-Stencil Pass is enabled in the project settings:
[image]

Ok, stay with me here. You need to activate Render "CustomDepth Pass" on the mesh you want to change the segmentation color on. Set the "CustomDepth Stencil Value" to the desired color in your palette. In the following screenshot it is set to seven:
[image]

I recommend changing some of CustomDepth Stencil Values to the same value and test if they have the same color in the segmentation view.

Regarding the palette:
It is a simple png file with 1024 pixels in width and 1 pixel in height. Every four pixels the color changes, thus you get 1024 / 4 = 256 different colors. Somehow colors on index 0 will not work the reason is somewhere in the bowels of the documentation dungeon. You can basically create the palette with any image manipulation program like GIMP or Paint etc.

I set the segmentation settings in the "settings.json" to:
```
   "SegmentationSettings":
  {
   "MeshNamingMethod": "OwnerName",
    "InitMethod": "None", 
    "OverrideExisting": false
  }
```

For me this worked for quite some time, but suddenly it stopped working and I cannot figure out why.
Please give me some feedback if this works for you!

sytelus, 2018-09-17:
This should work. The way to debug this is first use default pallet and make sure everything works. Then replace seg_color_pallet.uasset with your file. Make sure nothing else is changed. Also make sure other properties in .uasset is same as original .uasset. Also, view original .png and your .png side by side to make sure they have resolution. The code we use to generate png is [here](https://github.com/Microsoft/AirSim/blob/master/PythonClient/computer_vision/seg_pallete.py).

benjamin-lange, 2018-09-20:
Somehow the assigned CustomDepth Stencil Value gets overwritten.
I configured one of my models as follows:
[image]

When I run UE4 with Airsim and put out the CustomDepth Stencil Value for the model I get:
LogBlueprintUserMessages: [BP_Stockpile_1] 165

I found a workaround for this problem. In Blueprint on BeginPlay I assign the desired CustomDepth Stencil Value "again":
[image]

This time the segmentation color is correct in the segmentation view and the output is:
LogBlueprintUserMessages: [BP_Stockpile_1] 34

I am only using Blueprint in my project, thus this might be a possible bug in Airsim.
I tried removing the settings in the settings.json it made no difference at all.

benjamin-lange, 2018-09-20:
Turns out my hunch was correct. The Custom Depth Stencil Value was overridden at instance level. I had to go through the models and reset their values. I do not know why / how the values were overridden. But this issue is solved for me.

closed 2018-09-20 by benjamin-lange