microsoft/AirSim #1370: infrared camera?

koalamaster, 2018-09-04:
Is the Drone model in airsim available with an infrared camera?

sytelus, 2018-09-17:
This is work in progress. Currently there is [ImageType available called Infrared](https://github.com/Microsoft/AirSim/blob/127661aab68ce2da94ea069398c6f26eff312439/AirLib/include/common/ImageCaptureBase.hpp#L24) however it basically just sets the grey scale for segmentation view. So if you set right temperature per object, you might get "infrared-like" image but obviously there is a lot more to be done.

petergerten, 2019-05-22:
Is there currently any work in progress to improve the infrared view?

lucafei, 2021-01-08:
hi, does it have any information about infrared camera implementiert in custom environment?