microsoft/AirSim #1375: build fails

zaddan, 2018-09-05:
I have installed visual studio '17 and cloned the AirSim repo. However I get the following error when running build.cmd:
c1xx : fatal error C1083: Cannot open source file: 'common_utils\FileSystem.cpp': No such file or directory [C:\AirSim\MavLinkCom\MavLinkCom.vcxproj]

PS: I get plenty of errors similar to the above one complaining the compiler can not find the files.

zaddan, 2018-09-05:
This issue is resolved. I am not sure why this fixes the issue but if I clone using the "Bash on Ubuntu on Windows" app, this issue will occur but if I use "x64 Native Tools Command Prompt for VS 2017" to clone the repo, I won't have to deal with it.

sytelus, 2018-09-17:
Could you please tell us what is the sequence of command line you used?

closed 2018-12-18 by madratman