microsoft/AirSim #1377: Problems with distance units in moveToPositionAsync (drones)

saihv, 2018-09-05:
I am trying to use the `moveToPositionAsync` function to control multiple drones as shown in the multi_agent_drone.py file. Problem is, the distance units used in the arguments do not seems to correlate to the X, Y, Z position units used in settings.json. For instance, running these commands:  

```
f1 = client.moveToPositionAsync(0, 0, -8, 2, vehicle_name="Drone1")
f2 = client.moveToPositionAsync(1, 0, -8, 2, vehicle_name="Drone2")
f3 = client.moveToPositionAsync(2, 0, -8, 2, vehicle_name="Drone3")
```

I observe a lot more separation between the drones than 1 meter. How do I debug/fix this?

sytelus, 2018-09-17:
What positions are in your settings? One thing to understand is that each drone has its own local frame with origin as the starting point. So each drone starts at 0,0,0 in its local frame that maps to coordinates in world frame specified by settings.json. The world frame in turn has 0,0,0 at PlayerStart component in the UE environment.

saihv, 2018-09-17:
My settings.json asks for these starting positions:

```
      "Drone1": {
		  "VehicleType": "SimpleFlight",
		  "X": 0, "Y": 0, "Z": -4,
                 "Yaw": 90
		},
      "Drone2": {
		  "VehicleType": "SimpleFlight",
		  "X": 3, "Y": 0, "Z": -4,
                   "Yaw": 90
		},
	"Drone3": {
		  "VehicleType": "SimpleFlight",
		  "X": 6, "Y": 0, "Z": -4,
                 "Yaw": 90
       }
```

When I use the moveToPosition commands I listed in the question, the drones actually move **further** apart, instead of moving closer (because now the spacing should have reduced to 1 m between instead of 3), hence the confusion.

madratman, 2019-02-19:
I guess you figured this out already, but yes, they are moving wrt to their own frames

closed 2019-02-19 by madratman