microsoft/AirSim #1380: record segmentation images with reassigned object IDs

LunaXia, 2018-09-06:
Hello, I understand that using APIs in a script, I can get the desired segmentation view by assigning meshes with specific values and thereby changing the mesh colors. Using this approach, I have to add code to the script that saves the segmented images. Is there a way to modify the source code somewhere so that  I can use the record button to automatically save images with the desired segmentation view? Thank you!

benjamin-lange, 2018-09-11:
You can change the segmentation pallet material (this is mentioned in the docs) and assign CustomDepth Stencil Values on your meshes. Do not forget to enable Custom Depth-Stencil Pass in your project settings.
If it works for you let me know! Somehow my custom pallet is not working properly since my last update.

sytelus, 2018-09-17:
You can update [seg_color_pallet.png](https://github.com/Microsoft/AirSim/tree/master/Unreal/Plugins/AirSim/Content/HUDAssets) and then regenerate .uasset file for it. Here's [the code](https://github.com/Microsoft/AirSim/blob/master/PythonClient/computer_vision/seg_pallete.py) that we used to generate jpg file.

closed 2018-12-18 by madratman