microsoft/AirSim #1383: settings.json turns back to default.

WadeWuu, 2018-09-07:
I have already changed settings.json. But when I restart the project of UE, settings.json will  turns back to default. I can not understand.

WadeWuu, 2018-09-07:
I want to use multiple vehicles. So, I change setting.json like that.
{
	"SettingsVersion": 1.2,
	"SimMode": "Car",
	
	"Vehicles": {
		"Car1": {
		  "VehicleType": "PhysXCar",
		  "X": 4, "Y": 0, "Z": -2
		},
		"Car2": {
		  "VehicleType": "PhysXCar",
		  "X": -4, "Y": 0, "Z": -2,
      "Yaw": 90
		}
  }
}

But, when I start the project, it turns to the default.
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2
}

sytelus, 2018-09-17:
That's strange. Do you have non-english OS? I suspect there is a bug in code that checks if file already exist and it doesn't take in to account unicode characters.

madratman, 2018-12-18:
@WadeWuu any updates? was this resolved?

closed 2020-08-14 by jonyMarino