microsoft/AirSim #1386: Airsim features

Supaple-x, 2018-09-10:
Hey guys! 

I've found this(Airsim) miracle about a week ago! 
**Nice job!!!**
Would you mind me asking you about some features what I'm interested in:

- [ ] - is there a possibility to upload my 3d map into the engine?
- [ ] - is there a chance to upload custom models drones of mine?
- [ ] - is there a possibility to record and export the route of the drone in a sort of "flight mission" for a real drone in reality? And replay this recorded track in the engine
- [ ] - is there a chance to program the behavior of external moving objects? flying or moving on the ground?
- [ ] - what do you think if I can contribute in your project in practice? If I'm going to use this engine from model development, environment?
- [ ] - well and the final feature I want to ask about: can I use VR and consoles like PS4 and XBOX to manage vehicles in this airsim?

If this is acceptable it might be marvelous contribution project for me. I hope my activities might help you too.
If my questions have resolutions in your Airsim could you kindly give me the direction where to read "how to"s for making "my dreams" come true? :)

**Thanks a lot for this engine.
Waiting your soonest reply, gyus**

DavidLSmyth, 2018-09-10:
I can try and answer a few of these based on my interaction with AirSim so far:
- is there a possibility to upload my 3d map into the engine?
**Yes, Unreal engine will primarily support import in the .fbx autodesk format. If your 3D model can be converted to a .fbx file, import following this [guide](https://docs.unrealengine.com/en-us/Engine/Content/ImportingContent/ContentBrowserImport)**
- is there a chance to upload custom models drones of mine?
**Yes, see the documentation [here](https://github.com/Microsoft/AirSim/blob/master/docs/custom_drone.md)**
- is there a possibility to record and export the route of the drone in a sort of "flight mission" for a real drone in reality? And replay this recorded track in the engine
**Yes, see above documentation**
- is there a chance to program the behavior of external moving objects? flying or moving on the ground?
**Yes, for this you can use Unreal Engine's blueprint system. Here is an [example](https://www.unrealengine.com/en-US/blog/moving-physical-objects)**
- what do you think if I can contribute in your project in practice? If I'm going to use this engine from model development, environment?
**I'm not sure what you're asking here**
- well and the final feature I want to ask about: can I use VR and consoles like PS4 and XBOX to manage vehicles in this airsim?
**Your game should be easily ran in VR using unreal engine. Look at game packaging options if you want to create an PS4/XBox game (I think it would be a lot easier to upload to Steam). Documentation can be found [here](https://github.com/Microsoft/AirSim/blob/master/docs/remote_control.md) for connecting PS4/XBox controllers**

Supaple-x, 2018-09-11:
Thanks David!
A lot.
I will carefully revise this resources in a while.
**I'm not sure what you're asking here**
I ask if I can contribute in Airsim development with some my cases.
I would be happy if so. 
Well, thanks again David!
I'll try to build simulator...

closed 2018-09-26 by Supaple-x