microsoft/AirSim #1391: Initialize Twist and acceleration in the log file from the kinematics

geevargs, 2018-09-12:
I have the latest commit of airsim running on UE4 4.18 and on linux 16.04.
I am trying to get the twist and accelerations initialized from the kinematics.hpp and the references to https://github.com/Microsoft/AirSim/issues/487.
Dont seem valid to the present recording file.cpp.Is there a simpler way to log the angular velocity and the accelerations.

sytelus, 2018-09-17:
Could you please clarify what exactly are you doing and what you want. Try giving sample code, expected output, actual output etc.

geevargs, 2018-09-17:
I want to edit the code  so that I can get acceleration and angular velocity etc ,,, into /Documents/Airsim/Date/airsim_rec.txt

tejaswid, 2019-01-10:
+1
But since the IMU output is at a higher frequency than that of the images, it would be better to have it saved in a separate file. 
This would mean that the record button should launch another recording thread for the IMU data.

I have a python script (see below) that I launch separately when I start recording.
@sytelus Could you comment on this implementation - it would be nice if something like this is integrated into the record button with adjustable settings in the settings.json file.

```python
from threading import Thread
import setup_path
import airsim
import numpy as np

KEEP_SAVING=True

def save_imu(client, filename, hz):
    with open(filename, 'ab') as imu_file:
        while KEEP_SAVING:
            t = client.getMultirotorState().timestamp / 1e6
            lv = client.getMultirotorState().kinematics_estimated.linear_velocity
            la = client.getMultirotorState().kinematics_estimated.linear_acceleration
            aa = client.getMultirotorState().kinematics_estimated.angular_acceleration
            av = client.getMultirotorState().kinematics_estimated.angular_velocity
            imu = np.array([t, la.x_val, la.y_val, la.z_val, av.x_val, av.y_val, av.z_val])
            np.savetxt(imu_file, imu)
            # sleep to maintain required frequency
            time.sleep(1.0/hz)

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()

thread = Thread(target=save_imu, args=(client,"output.dat", 100))
thread.start()

airsim.wait_key("Press any key to stop recording.")
KEEP_SAVING = False
```

pengjili, 2019-02-18:
+1
Want to see them in airsim_rec.txt

pengjili, 2019-02-21:
@madratman Since in real-world autonomous driving, the IMU frequency is usually much higher than camera's frequency. If IMU or GPS data can be outputted based on its own frequency, that would be so great. Hope to see this kind of APIs appear in future version.

madratman, 2019-04-30:
Hey @pengjili, I've added the IMU API in https://github.com/Microsoft/AirSim/pull/1920, however it's not in the recording / separate thread . 
If using it via the C++ / python API  suits your needs, please close this.

closed 2021-01-14 by jonyMarino