microsoft/AirSim #1393: Can I draw the line when I run the UE4

EricJin101, 2018-09-13:
[image]
As the picture shows...

EricJin101, 2018-09-13:
I need to display the global path in UE, how can I make it?

DavidLSmyth, 2018-09-13:
If you edit BP_FlyingPawn to include a draw debug line node you should get the result shown in the picture. Here is what I used in my project: 
[image]

EricJin101, 2018-09-17:
> If you edit BP_FlyingPawn to include a draw debug line node you should get the result shown in the picture. Here is what I used in my project:
> [image]

What I really wanted to show was that once I had the planned trajectory I could visualize it on the AirSim directly, similar to the way one can in rviz.

DavidLSmyth, 2018-09-17:
Your question seems pretty vague. If you want to view a planned trajectory you could just run the simulation with the clock significantly sped up?  As far as I know there is no way to visualise a planned trajectory without running the game as the simulator will start when the game runs, and the simulator is what estimates the state of the vehicles (simulator code [here](https://github.com/Microsoft/AirSim/blob/a4fcf568e75b6d675d5d127754757d673ce33ecb/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleFlightEstimator.hpp))

DavidLSmyth, 2018-10-16:
I updated the api to visualise planned waypoints. I made the changes to worldSimApi, similar to how the printLogMessage functionality was added. I can submit a pull request if this would be useful to add to the api.
[image]

EricJin101, 2018-10-16:
thank you! it will be extremely useful! I believe it is the exactly API I want.

EricJin101, 2018-10-17:
> I updated the api to visualise planned waypoints. I made the changes to worldSimApi, similar to how the printLogMessage functionality was added. I can submit a pull request if this would be useful to add to the api.
> [image]

thanks a lot! It's really useful

DavidLSmyth, 2018-10-17:
I have it in my fork: https://github.com/DavidLSmyth/AirSim/
Changes are:

- You can now draw debug lines relative to a vehicle or to the 0,0,0 coordinate in the environment
- There is an option to turn a debug tracing line on for the RAV which tracks its path. 

Examples can be found in takeoff.py, new methods are showPlannedWaypoints and simShowDebugLines.

It's slightly messy at the moment as I had some trouble figuring out how to use Vector3r with RPC but I will fix that soon.

EricJin101, 2018-10-19:
> I have it in my fork: https://github.com/DavidLSmyth/AirSim/
> Changes are:
>
>     * You can now draw debug lines relative to a vehicle or to the 0,0,0 coordinate in the environment
>
>     * There is an [...]

I got an error...
_/vehicles/multirotor/api/MultirotorApiBase.cpp:7:10: fatal error: 
      'DrawDebugHelpers.h' file not found
#include "DrawDebugHelpers.h"_

DavidLSmyth, 2018-10-19:
What unreal engine version and OS are you using?

EricJin101, 2018-10-22:
> What unreal engine version and OS are you using? It's included in multiple files that existed before I made changes so strange that it's only showing that error now
> [image]
> 
> I'm developing on windows 10, visual studio 2017, unreal engine 4.18 so there may be compatibility issues.

sorry for the reply is not timely, I'm developing on Ubuntu 16.04, and UE is 4.20, my computer broke down last weekend... 
I will try it again this week

ZWYDeveloper, 2018-10-24:
> I got an error...
> _/vehicles/multirotor/api/MultirotorApiBase.cpp:7:10: fatal error: 'DrawDebugHelpers.h' file not found #include "DrawDebugHelpers.h"_
Hello, I have encountered this problem, have you solved it?

ZWYDeveloper, 2018-10-24:
Where is this file "drawdebughelpers.h" from?UE4?So how do you compile airsim?

EricJin101, 2018-10-24:
@ZWYDeveloper I am on a business trip, sadly, I didn't figure it out yet..

DavidLSmyth, 2018-10-24:
What compiler are you using? I have ubuntu dual partitioned so I'll try it and post a fix. I imagine it's an issue with the compiler. DrawDebugHelpers is from UE, provides debugging visualisation functionality

ZWYDeveloper, 2018-10-25:
@DavidLSmyth Yes, I succeeded, I send this file before compilation airsim commented out, put the compiled plug-in ue4 compilation in the scene and then opens, the effect of this is I compiled, is nice, but I want to optimize the trajectory of the lines, I need to take a look at you to modify the API, thank you for your open source share, thank you!

ZWYDeveloper, 2018-10-25:
[image]
[image]

DavidLSmyth, 2018-10-25:
> @DavidLSmyth Yes, I succeeded, I send this file before compilation airsim commented out, put the compiled plug-in ue4 compilation in the scene and then opens, the effect of this is I compiled, is [...]

The changes were pretty straightforward, they are all contained in one commit. Happy to help!

RVBldr, 2018-11-21:
Internal VSTF 7413.

AloshkaD, 2018-12-26:
@DavidLSmyth  Thank you for sharing your addition to path plotting. I had an error with`self.simShowPawnPath(True, 3, 8,vehicle_name)`
as follows 
`msgpackrpc.error.RPCError: rpclib: server could not find function 'simShowPawnPath' with argument count 4.`
I have installed your airsim fork to make sure I don't miss anything. 
Also do I still need to edit BP_FlyingPawn ?

DavidLSmyth, 2018-12-26:
> @DavidLSmyth Thank you for sharing your addition to path plotting. I had an error with`self.simShowPawnPath(True, 3, 8,vehicle_name)`
> as follows
> `msgpackrpc.error.RPCError: rpclib: server could not [...]

I made some changes recently that may have broken some things, I'll have a look and get back to you when I've figured out what's gone wrong. If you don't need some of the latest functionality you could use one of the older commits (try 129596f7e964d96936b46a54c89b44f6a36bf57c)

AloshkaD, 2018-12-28:
Thanks @DavidLSmyth  I look forward to that. Regarding the commit that you referenced, it does not seem related. It's an update to the  simGetPositionWRTOrigin method.

Hoshea7, 2019-07-28:
Hi @AloshkaD ,

I still cannot figure out how to draw the line of the trajectory of multirotor, could you kindly explain more how you complete this ?

tteague19, 2021-12-03:
@DavidLSmyth I am encountering an error that "rpclib: server could not find function 'simShowDebugLines' with argument count 0" when I try to use your fork. Strangely, the function is defined in your .hpp and .cpp files with the appropriate number of arguments. Would you be able to help with this issue? Thanks!

DavidLSmyth, 2021-12-06:
I'll see if I can find the time to submit this as a clean pull request. Haven't done any maintenance on this fork for years...

tteague19, 2021-12-08:
Thanks, @DavidLSmyth! In case you are unable to make the fix, do you know if there has been a similar capability developed within AirSim? Essentially, I want to indicate a connection between two or more multirotors for the purposes of visualization.

DavidLSmyth, 2021-12-08:
> Thanks, @DavidLSmyth! In case you are unable to make the fix, do you know if there has been a similar capability developed within AirSim? Essentially, I want to indicate a connection between two or more multirotors for the purposes of visualization.

I don't think so (you can turn on debug lines showing where the drone has flown, that's the closest thing I'm aware of) but you could put this together using a blueprint rather than making a change to the API (if you've seen images of other people doing this I'd guess they've just used a blueprint). I got pretty frustrated with blueprints in the end so I'd recommend to do this from the API if possible!

DavidLSmyth, 2021-12-09:
> Thanks, @DavidLSmyth! In case you are unable to make the fix, do you know if there has been a similar capability developed within AirSim? Essentially, I want to indicate a connection between two or more multirotors for the purposes of visualization.

Was just going through it there, maybe [simPlotLineList](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simPlotLineList) is what you want? I wasn't aware this had been added!

tteague19, 2021-12-11:
It is! By coincidence, I found that utility earlier this week, as well. Ideally, I would plot the lines throughout the simulation, but it suffices to use it between movements, which seems to be the only way to do so. If I become more proficient with AirSim, I will consider implementing something closer to what I have in mind. Thank you again for your help, @DavidLSmyth!