microsoft/AirSim #1394: Question: drone starts to shake if the altitude is over 50m

xiahaa, 2018-09-13:
Hi, 
Version: Airsim 1.2.0, binary.
Control: call moveOnPathSync with default setting but assigned waypoints.
Problem: looks fine if the altitude is lower than 50m. However, if I tried to fly along the same path with an altitude over 60m, the drone started to shake. Does anyone have an idea about this abnormal phenomenon?

xiahaa, 2018-09-14:
> I faced the same problem with moveToPositionAsync() when the drone flies over or around a certain height. But I have no idea about this.

It seems quite strange. Could the aerial dynamics constraints be the reason?
I added gimbal, so the footage you will get is ok. But the drone's attitude is a mass...

sytelus, 2018-09-17:
Could you please provide the sample code to reproduce this? I'm guessing your speed is too high.

xiahaa, 2018-09-18:
> Could you please provide the sample code to reproduce this? I'm guessing your speed is too high.

```
sss
```

madratman, 2018-12-19:
@xiahaa can you please try the branch [bug_fixes/z_controller](https://github.com/Microsoft/AirSim/tree/bug_fixes/z_controller) and get back to us?

xiahaa, 2018-12-21:
> @xiahaa can you please try the branch [bug_fixes/z_controller](https://github.com/Microsoft/AirSim/tree/bug_fixes/z_controller) and get back to us?

Sure~

madratman, 2019-01-08:
@xiahaa any news?

hoangtranngoc, 2019-01-11:
Hi @madratman! 
In my case, the error is still there. The drone is still shaking. 
You can see my recorded video here https://youtu.be/VOaxnu0XI_s

xiahaa, 2019-01-12:
> @xiahaa any news?

Sorry for the delay.
I have tried with the branch you provided. The behavior is even more wired, from what I have seen~
I made a gif that you can see the behaviors.

[image]

stavBodik, 2019-04-18:
Any solution for this ? Its looks like Weather : Wind bug.

ACLeighner, 2020-05-13:
Still getting this bug in the latest AirSimNH environment

ACLeighner, 2020-05-18:
Ok, I think I found a short term fix, although looking at the code it looks like someone had something more elegant planned but it never got finished.

In MultirotorAPIBase.cpp I changed some code in the moveToPathPosition 
 Function:

Replace
```c++
//send commands
//try to maintain altitude if path was in XY plan only, velocity based control is not as good
    if (std::abs(cur.z() - dest.z()) <= 1) //for paths in XY plan current code leaves z untouched, so we can compare with strict equality
        moveByVelocityZInternal(velocity_vect.x(), velocity_vect.y(), dest.z(), yaw_mode);
    else
        moveByVelocityInternal(velocity_vect.x(), velocity_vect.y(), velocity_vect.z(), yaw_mode);```

with

```c++
//send commands
moveByVelocityInternal(velocity_vect.x(), velocity_vect.y(), velocity_vect.z(), yaw_mode);```

fawadahm, 2020-06-15:
This did not help at higher altitudes. We still see a lot of jitter/vibrations at 70m/80m heights. @madratman, are there any other fixes we can try?

ACLeighner, 2020-06-18:
Huh, you rebuilt with that change and no luck?  I just tried flying at 90m and didnt have any issues...If someone else wants to try, it looks like the issue is is moveByVelocityZInternal, which causes the drone to very quickly change its zvelocity  at the target altitude.  So it is essentially stuck oscillating (shaking) at that altitude.

ACLeighner, 2020-06-19:
OK! I realize now that my earlier solution doesn't work at higher velocity...Edited my earlier comment with a new change that seems to be working

fawadahm, 2020-06-22:
This works. Thank you!

lucafei, 2020-10-31:
i have alse met this problem? do you guys have any solution for that?

nethtrick, 2021-02-16:
Can you verify this is not an artifact of the Blocks environment? I believe at one point there was an invisible "ceiling" in that level,

PR #3478 (2021-03-17): Fix drone shaking at high altitudes

Fixes: #1394
Fixes: #2802

## About
Updates MultirotorApiBase.cpp to use the velocity PID with target z velocity 0 when we are close to our goal altitude to fix drone shaking. Thank you @ACLeighner for finding this fix!

## How Has This Been Tested?
A test script was used that calls moveToPositionAsync to send the drone to an altitude of 60m then flies the drone in a square path to observe the shaking described in #1394.

## Screenshots (if appropriate):
closed 2021-03-17 by zimmy87