microsoft/AirSim #1396: px4

geevargs, 2018-09-14:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----

asimukaye, 2018-09-19:
I am facing a similar error. Specifically, the following is the description of the error(s):
1. The baro and GPS altitude does not change after takeoff. This is causing the client.takeoffAsync() call to make the multirotor to go up infinitely.
2. The GPS location gets reset to the home location after a few seconds of manual flight.
3. This problem takes place in both HILS and SITL mode (with the corresponding settings as prescribed in settings.json)
4. I cannot fly the multirotor in position or mission modes as a result as it returns a critical navigation failure error.
Note: This problem did not occur with Airsim 1.1.10; 
Note the baro and GPS altitudes in the two screenshots; one on ground and the other in flight.
[image]
[image]

HoaPham2711, 2018-09-19:
I am facing this problem too.

HoaPham2711, 2018-09-20:
I was able to fix this problem. Please re-clone PX4 Firmware (https://github.com/PX4/Firmware.git)
then **checkout tags/v1.7.0** using 

> git checkout tags/v1.7.0 && git submodule update --init --recursive

HoaPham2711, 2018-09-21:
Sorry, my solution is not radical. The drone still takes off to infinity :(

mohammedshafeeqet, 2018-09-24:
@lovettchris Some help here, please.

geevargs, 2018-09-25:
I was able to get it working...using unreal 4.17.2 and Airsim unreal4.16...with pixhawk firmware 1.6.3

mohammedshafeeqet, 2018-09-26:
@geevargs Can you please confirm if it is Airsim v1.2 with Unreal 4.17.2?

ericamoroso, 2018-09-26:
Had the same issue. Looks like the "Environment" class isn't getting properly handled between UE4 and AirLib. A hack "solution" for Airsim v1.2 is to uncomment lines 115 and 116 within AirLib/include/physics/PhysicsBody.hpp.

HoaPham2711, 2018-09-27:
Thank you @ericamoroso . Your solution worked for me

geevargs, 2018-09-27:
@mohammedshafeeqet  https://github.com/Microsoft/AirSim/tree/Unreal4.16...is what I used

asimukaye, 2018-10-01:
Thanks @ericamoroso , your solution works for me as well

lovettchris, 2018-12-18:
A bunch of bugs have been fixed in AirSim and PX4 SITL model should work a lot better now.

Planche95, 2018-12-31:
I had that problem on release v1.2.1 - Windows, but when I build AirSim from actual master branch everything worked fine.

lovettchris, 2018-12-31:
great to hear, thanks for trying it.

Zachmoist, 2019-09-17:
Hello, I'm currently having the same issue, i'm using Windows 10 and Airsim 1.2.0, PX4 1.8.2, QGroundcontrol v3.4.3 and Python 3.6

My settings.json:
`{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false
        }
    }
}`

When use QGC and make a move to position i get the navigation error.
[image]

If i don't use QGC and use 'commander arm' and 'commander takeoff' the drone starts to fly in an orbit until it starts collisioning to the ground, i try to get the home position but the 'pos' command doesn't work for me, i get a "INFO [gps] not running" message.

Also, i never get the home position log message, i start the PX4 and wait for around 15 seconds but the most i get is this:
[image]

ironclownfish, 2020-08-13:
This appears to behave better with AirSim v1.2.1 and newer. @Zachomist please try updating beyond v1.2.0 and open a new issue if the problem persists for you (since that would imply it's not actually fixed in newer versions as Planche95 suggests.) Thanks

closed 2020-08-13 by ironclownfish