microsoft/AirSim #1397: Question: How to get car state in C++?

dehavenm, 2018-09-14:
I'm working on a C++ ROS node to interface a ROS package with a car in Airsim. To properly control the car, I would need to be able to read the car's speed and angular velocity, then compare them to the commanded values coming through the ROS messages. The example code mentions a client.getCarState() command for Python, but nothing in the documentation mentions anything equivalent for C++. How do I get the car's state through C++? Thank you.

sytelus, 2018-09-17:
For C++ you use CarRpcLibClient. Method is 

`CarApiBase::CarState getCarState(const std::string& vehicle_name = "")`

APIs: https://github.com/Microsoft/AirSim/blob/77f84fff2c60d9b3d361bc2d2121cf10436d0996/AirLib/include/vehicles/car/api/CarRpcLibClient.hpp#L19

Example: https://github.com/Microsoft/AirSim/blob/77f84fff2c60d9b3d361bc2d2121cf10436d0996/HelloCar/main.cpp#L19

dehavenm, 2018-09-17:
No method by the name getCarState appears in either of those files.

bimalkmehta, 2018-12-18:
@dehavenm - were you able to use the API (it is defined).. tho there doesn't seem to be a sample for it quite as yet.  Perhaps you can post it, if you managed to work it out.

jonyMarino, 2020-08-12:
@dehavenm did you find how to use getCarState() in C++?

dehavenm, 2020-08-12:
No, I never was able to find that information.

jonyMarino, 2020-09-17:
Hi @dehavenm! Yo can see that method in the file commented by stylus but three lines below. I tried adding this line `CarApiBase::CarState state = client.getCarState();` to HelloCar/main.cpp successfully. Let me know if you need more help. Regards

closed 2020-09-17 by jonyMarino