microsoft/AirSim #1403: Magnetometer model

mouchuanlin, 2018-09-17:
Hi,
I am working to use a different magnetometer for AirSim. In reviewing the MagnetometerSimpleParams.hpp, there are couple of struct members I don't have much info. 
Here they are
- scale_factor = 1.0 (what's this factor for?)
- zero gauss level (I can't find this in the MMC5883MA datasheet)
- ref_update_frequency = 0.2f (what's this for?)

Does anyone has some insight for how to create a new magnetometer model?

Thx

mouchuanlin, 2018-09-17:
I also found there is a scale_factor value changed from 2.0f to 1.0f. What's the change for?
Change magnetometer scale to match hmc5883 sensor.

sytelus, 2018-10-17:
If I remember correctly, scale_factor can be used to simulate imperfect calibration. The ref update frequency is how fast do we update magnetic field ground truth wrt to position. This computation is pretty expensive but magnetic field barely changes on sphere of size of Earth if you are moving only few 100s of meters. So we really don't need to keep doing this computation every physics frame (which runs at 100+ Hz).

closed 2021-01-14 by jonyMarino