microsoft/AirSim #1404: .confirmConnection() error

lloydrayner, 2018-09-17:
Hi All,

hoping for some help. I am trying to run Airsim through python 3.7.0 on a Mac. I have UE4 open but keep getting the following error when running the 'client.confirmConnection()' command. 

Can anyone help?

`WARNING:tornado.general:Connect error on fd 23: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 13: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 13: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 13: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 13: ECONNREFUSED
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/lloyd/.virtualenvs/Fly/lib/python3.7/site-packages/airsim/client.py", line 54, in confirmConnection
    if self.ping():
  File "/Users/lloyd/.virtualenvs/Fly/lib/python3.7/site-packages/airsim/client.py", line 24, in ping
    return self.client.call('ping')
  File "/Users/lloyd/.virtualenvs/Fly/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/Users/lloyd/.virtualenvs/Fly/lib/python3.7/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit`

MaximStrnad, 2018-09-20:
Start the python client before selecting a vehicle mode.

lloydrayner, 2018-09-29:
probably a silly questions but how do I start the python client?

MaximStrnad, 2018-10-01:
Install python with Miniconda. Both windows and linux. Then navigate into the python directory. (most likely AirSim/PythonClient/Multirorot and run a command line there. Then just simply start the python with command `python hello_drone.py`. That should do it. If you dont know how to setup the unreal engine enviroment I suggest you should search for a video on youtube. I saw there was one that showcased the way to install unreal as plugin into your own projects. If you have any more questions feel free to ask.

closed 2019-02-19 by madratman

aoltonen, 2019-04-09:
Hi~ I met with the same problem when running the DQNdrone.py in the spyder. So I wonder how you finally solved the problem?