microsoft/AirSim #1413: Blocks env fails: NullSourceCodeAcessor not found

stratomaster31, 2018-09-21:
When I try to open the blocks project from the UE4editor i get this prompt error messahe:

[image]

Any help? Thx in advance

anphunl, 2018-10-09:
Same issue here.

[image]

Operating System: Ubuntu 16.04 64bit
CPU: Intel Core i5 8400
GPU: Nvidia GTX 1080 Ti
RAM: 32 GB
Flight Controller: None
Remote Control: None

sytelus, 2018-10-17:
I think there is an issue with your install. See here: https://answers.unrealengine.com/questions/562619/unreal-engine-414-linux-no-compiler-was-found.html

RVBldr, 2018-12-07:
@stratomaster31 Is your issue resolved?

ssokhey, 2019-01-24:
@stratomaster31 @anphunl @RVBldr 
There are multiple things you have to check to make it work for example your clang version, plugins folder etc.

Better to use newly released blocks binary. [https://github.com/Microsoft/AirSim/releases](url)

jonyMarino, 2020-08-14:
closed due to age

closed 2020-08-14 by jonyMarino