microsoft/AirSim #1414: The RecordInterval Problem

jinlukang1, 2018-09-22:
I set my RecordInterval:6.00, but when I check my AirSim File, I found that 8 images were recorded in one second of 6 seconds, and when I ran again, it became 4 images, and sometimes 1 and sometimes 2, why?

this is my setting.json
```
{
	"SettingsVersion": 1.2,
	"SimMode": "ComputerVision",
	"ViewMode": "",
	"Recording": {
		"RecordOnMove": false,
		"RecordInterval": 6.00,
		"Cameras": [
			{ "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
			{ "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
		]
	},
	"CameraDefaults": {
	"CaptureSettings": [
		{
		  "ImageType": 0,
		  "Width": 1440,
		  "Height": 900,
		  "FOV_Degrees": 90,
		  "AutoExposureSpeed": 100,
		  "MotionBlurAmount": 0
		},
		{
		  "ImageType": 5,
		  "Width": 1440,
		  "Height": 900,
		  "FOV_Degrees": 90,
		  "AutoExposureSpeed": 100,
		  "MotionBlurAmount": 0
		}
	]
	},
	"SegmentationSettings": {
	"InitMethod": "CommonObjectsRandomIDs",
	"MeshNamingMethod": "OwnerName",
	"OverrideExisting": false
  }
}
```

Can Someone help me?

sytelus, 2018-10-17:
Just tested this and it works as expected. At each interval one image for each camera is captured.

Relevant code: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Recording/RecordingThread.cpp#L81

madratman, 2018-12-07:
@jinlukang1 was this resolved?

jinlukang1, 2018-12-13:
No, I don't know how to solve this problem yet.
So I just close it after I have recorded it So when I run it again, it's right again.

royuqp, 2019-06-26:
Hi, I found that this problem occurs when you begin play more than once on the editor and then start recording. Each time you begin play, "Cameras" setting of "Recording" written in settings.json is internally accumulated and used for recording.

The following simple source code modification will fix it :)

`static void loadRecordingSetting(const Settings& settings_json, RecordingSetting& recording_setting)`
`{`
`    recording_setting.requests.clear();    // <- inserted`
`    Settings recording_json;`
`...`