microsoft/AirSim #1415: problem on additional camera setting

hashikemu, 2018-09-22:
Hi, I use airsim for research. I want to add camera or change camera position and orientation.
So, i wrote this setting.json.
```
{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	"SettingsVersion": 1.2,
	"SimMode": "Car",
	"ClockSpeed": 1.0,
	"SubWindows": [
	{"WindowID": 0, "ImageType": 0, "CameraName": "MyCamera2", "Visible": true},
	{"WindowID": 1, "ImageType": 0, "CameraName": "MyCamera1", "Visible": true},
	{"WindowID": 2, "ImageType": 0, "CameraName": "MyCamera3", "Visible": true}
	],
	"Vehicles": {
		"Car1": {
			"VehicleType": "PhysXCar",
			"Cameras": {
				"MyCamera1": {
					"CaptureSettings": [
						{
							"ImageType": 1,
							"Width": 256,
							"Height": 144,
							"FOV_Degrees": 100
						}
					],
					"X": 1.87, "Y": 0, "Z": -1.1,
					"Pitch": -15, "Roll": 0, "Yaw": 0
				},
				"MyCamera2": {
					"CaptureSettings": [
						{
							"ImageType": 1,
							"Width": 256,
							"Height": 144,
							"FOV_Degrees": 100
						}
					],
					"X": 0, "Y": -0.55, "Z": -1.6,
					"Pitch": -15, "Roll": 0, "Yaw": -90
				},
				"MyCamera3": {
					"CaptureSettings": [
						{
							"ImageType": 1,
							"Width": 256,
							"Height": 144,
							"FOV_Degrees": 100
						}
					],
					"X": 0, "Y": 0.55, "Z": -1.6,
					"Pitch": -15, "Roll": 0, "Yaw": 90
				},
				"MyCamera4": {
					"CaptureSettings": [
						{
							"ImageType": 1,
							"Width": 256,
							"Height": 144,
							"FOV_Degrees": 100
						}
					],
					"X": -1.97, "Y": 0, "Z": -1.1,
					"Pitch": -15, "Roll": 0, "Yaw": 180
				}
			},
			"X": 707, "Y": 568, "Z": -2,
			"Pitch": 0, "Roll": 0, "Yaw": 0
		}
	}
}
```

And, I overwrite line 56 in hello_car.py. like this.
```
airsim.ImageRequest("1",...)
↓
airsim.ImageRequest("MyCamera1",...)
```

But, in simGetImages function, exception error occurred. error message is below.
```
Traceback (most recent call last):
  File "C:\Users\hashimoto\altGit\AirSim\PythonClient\car\hello_car.py", line 56, in <module>
    airsim.ImageRequest("MyCamera1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGBA array
  File "C:\Users\hashimoto\altGit\AirSim\PythonClient\airsim\client.py", line 93, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name)
  File "C:\Users\hashimoto\AppData\Local\Programs\Python\Python36\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\hashimoto\AppData\Local\Programs\Python\Python36\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'simGetImages' (called with 2 arg(s)) threw an exception. The exception contained this information: invalid map<K, T> key.
```

Since settings are reflected in subwindow, I guess setting.json is not wrong.
Also, I am thinking from the error message that types.py lack of function itself.
Is there anyone who added a camera and did it well?

michael-fonder, 2018-10-04:
Hi,
I'm facing the exact same issue... Did you find what was wrong?

sytelus, 2018-10-17:
Just tried out with latest PIP client (1.2.3) and did not see any issues when using same settings you have specified and making code change as you have suggested. Here's my output:

```
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Speed 0, Gear 0
Go Forward
Go Forward, steer right
Go reverse, steer right
Apply break
Retrieved images: %d 4
```

zoeeer, 2018-11-12:
Hi guys, any update on this?

I have the exact problem - my camera settings (looks pretty much the same as @hashikemu 's) are *almost* correctly reflectly in subwindows, but simGetImage api failed with the same error.
(I'll explain the *almost* part later)

@sytelus i'm not sure what you mean by PIP client version 1.2.3, could you explain a little? I'm new to AirSim and UE, and I guess it's not the python package manager pip, right? How do I know my PIP version and how to upgrade it if its not the latest?

I'm running the CityEnviron in v1.2.0 Windows release.

#### More about camera settings (what I meant by *almost* correct):
There're two parts of settings related to cameras: "CameraDefaults" and "Cameras" in a specific vehicle.

  - Some changes I made in "CameraDefaults" are applied to "MyCamera" added in my vehicle, but are not applied to the pre-defined cameras (eg. "0" or "front_center") such as the exposure settings
  - Other changes I made in "CameraDefaults" applied to the pre-defined cameras but not "MyCamera", such as the resolution settings.

These are just not what you would expect until you try them out... Is this a bug or is it by design?

hashikemu, 2018-11-12:
I'm sorry I did not reply for a long time.
1.I found a solution to edit CarPawn.cpp
2.As @sytelus says, by updating AirSim's Pythonclient and UnrealEngine's plugins, you can get images from additional cameras.
I will explain 1 and 2 below.

1.Solution by editing Carpawn.cpp
Rewrite the camera part of Carpawn.cpp with the following code.
```
// Create In-Car camera component
camera_front_center_base_ = CreateDefaultSubobject<USceneComponent>(TEXT("camera_front_center_base_"));
// camera_front_center_base_->SetRelativeLocation(FVector(200, 0, 100)); //center
camera_front_center_base_->SetRelativeLocation(FVector(187, 0, 110)); //center
camera_front_center_base_->SetRelativeRotation(FRotator(-15.0f, 0.0f, 0.0f)); //center
camera_front_center_base_->SetupAttachment(GetMesh());
camera_front_left_base_ = CreateDefaultSubobject<USceneComponent>(TEXT("camera_front_left_base_"));
// camera_front_left_base_->SetRelativeLocation(FVector(200, -12.5, 100)); //left
camera_front_left_base_->SetRelativeLocation(FVector(0, 55, 160)); //left
camera_front_left_base_->SetRelativeRotation(FRotator(-15.0f, 90.0f, 0.0f)); //left
camera_front_left_base_->SetupAttachment(GetMesh());
camera_front_right_base_ = CreateDefaultSubobject<USceneComponent>(TEXT("camera_front_right_base_"));
// camera_front_right_base_->SetRelativeLocation(FVector(200, 12.5, 100)); //right
camera_front_right_base_->SetRelativeLocation(FVector(0, -55, 160)); //right
camera_front_right_base_->SetRelativeRotation(FRotator(-15.0f, -90.0f, 0.0f)); //right
camera_front_right_base_->SetupAttachment(GetMesh());
camera_driver_base_ = CreateDefaultSubobject<USceneComponent>(TEXT("camera_driver_base_"));
camera_driver_base_->SetRelativeLocation(FVector(0, -25, 125)); //driver
camera_driver_base_->SetupAttachment(GetMesh());
camera_back_center_base_ = CreateDefaultSubobject<USceneComponent>(TEXT("camera_back_center_base_"));
// camera_back_center_base_->SetRelativeLocation(FVector(-200, 0, 100)); //rear
camera_back_center_base_->SetRelativeLocation(FVector(-197, 0, 110)); //rear
camera_back_center_base_->SetRelativeRotation(FRotator(15.0f, 0.0f, 0.0f)); //rear
camera_back_center_base_->SetupAttachment(GetMesh());
```
You can change the default camera placement by compiling and running it. I do not recommend this method because the brute force.

2.Solution by updating
I first rewrote the code because I needed to solve it by the first method, but then I solved it the second way.
It solved it by updating "both" of Pythonclient and UnrealEngine plugin.
It seems that it will not work just by updating Pythonclient.
By the way, updating from pip is possible by attaching the -U option.
`pip install airsim -U`

zoeeer, 2018-11-12:
hi @hashikemu ,

v1.2.1 works great! Thanks for your thorough reply.
I saw v1.2.1 there earlier today, but did not try it 'cause it's marked as pre-release.

However the other behavior i mentioned still exists, i.e. the "CameraDefaults" problem. I can make my desired settings in the "Vehicles" part, though.

madratman, 2018-12-19:
@sytelus I faced the exact same issue mentioned by @zoeeer above in the hackathon.. `TargetGamma` did the job for changing the exposure. not sure what AutoExposureMaxBrightness or AutoExposureSpeed do?

cacadam89, 2019-04-26:
I am seeing something very similar where I cannot access the camera images that I specify in my settings.json file, except now this issue is only occuring in ComputerVision mode. Below are my two setting files, and I am able to access the images from the first one that is spawning a vehicle but NOT from the one using ComputerVision mode. 

I am accessing currently using airsim with ros, and through rostopic echo I confirmed there was still image data being published in the ComputerVision case, so I tried the default camera `responses = client.simGetImages(["1", airsim.ImageType.Scene, False, False)])`  with the default image size of 144x256 and this _did_ show an image, but this image is from a default camera and not one controlled by my settings file.

**Working** settings.json file
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "scene_gt", "Visible": true},
    {"WindowID": 1, "ImageType": 2, "CameraName": "depth_gt", "Visible": true},
    {"WindowID": 2, "ImageType": 2, "CameraName": "depth_noisy", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 2,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "scene_gt": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": 0.00, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "depth_noisy": {
          "CaptureSettings": [
            {
              "ImageType": 2,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
              "Enabled": true,
              "ImageType": 2,
              "RandContrib": 0.2,
              "RandSpeed": 100000.0,
              "RandSize": 500.0,
              "RandDensity": 2,
              "HorzWaveContrib":0.03,         
              "HorzWaveStrength": 0.08,
              "HorzWaveVertSize": 1.0,
              "HorzWaveScreenSize": 1.0,
              "HorzNoiseLinesContrib": 1.0,
              "HorzNoiseLinesDensityY": 0.01,
              "HorzNoiseLinesDensityXY": 0.5,
              "HorzDistortionContrib": 1.0,
              "HorzDistortionStrength": 0.002
            }
          ],
          "X": 0.50, "Y": 0.00, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "depth_gt": {
          "CaptureSettings": [
            {
              "ImageType": 2,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": 0.00, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}`

**Not working** file:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "Cameras": {
    "scene_gt": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 672,
          "Height": 376,
          "FOV_Degrees": 90
        }
      ],
      "X": 0.00, "Y": 0.00, "Z": 0.00,
      "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
    },
    "depth_gt": {
      "CaptureSettings": [
        {
          "ImageType": 2,
          "Width": 672,
          "Height": 376,
          "FOV_Degrees": 90
        }
      ],
      "X": 0.00, "Y": 0.00, "Z": 0.00,
      "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
    },
    "depth_noisy": {
      "CaptureSettings": [
        {
          "ImageType": 2,
          "Width": 672,
          "Height": 376,
          "FOV_Degrees": 90
        }
      ],
      "NoiseSettings": [
        {
          "Enabled": true,
          "ImageType": 2,
          "RandContrib": 0.2,
          "RandSpeed": 100000.0,
          "RandSize": 500.0,
          "RandDensity": 2,
          "HorzWaveContrib":0.03,         
          "HorzWaveStrength": 0.08,
          "HorzWaveVertSize": 1.0,
          "HorzWaveScreenSize": 1.0,
          "HorzNoiseLinesContrib": 1.0,
          "HorzNoiseLinesDensityY": 0.01,
          "HorzNoiseLinesDensityXY": 0.5,
          "HorzDistortionContrib": 1.0,
          "HorzDistortionStrength": 0.002
        }
      ],
      "X": 0.00, "Y": 0.00, "Z": 0.00,
      "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
    }
  }
}`

ironclownfish, 2020-08-13:
The original question was resolved by Sytelus and Hashikemu when they discussed updating airsim, so this is closed.
@cacadam89 please feel free to open a separate issue for your question!

closed 2020-08-13 by ironclownfish