microsoft/AirSim #1416: Custom AirSim version for research purposes

eds89, 2018-09-23:
Hello,
As part of a research project, I plan to extend AirSim. Do I need to license a map (e.g. Neighbourhood) in order to include it on a custom distribution? The extended version will be not for profit and purely for research purposes. The API features already included may not be sufficient, hence the need for changing its source directly.

eds89, 2018-09-23:
> In general, when you purchase 3d content for making custom UE4 maps, or
> extending existing maps, the content is licensed for distribution of UE4
> binaries, but you can't redistribute the source content. Microsoft can't
> redistribute editable version of maps that contain content it doesn't own
> unlimited distribution rights to.

Where and how did you find the Neighbourhood map? I could not find it on
the Unreal Marketplace. Also, where do the car models come from?

TritonSailor, 2018-09-23:
I think this is the starting point for the Neighborhood environment.

https://www.unrealengine.com/marketplace/modular-neighborhood-pack

Car models can be either props (static meshes) that don't move, or rigged vehicles (skeletal meshes) that can be setup as car pawns.  You can find both types on the marketplace, or make your own.

eds89, 2018-09-23:
Did the AirSim project buy these packs?

RVBldr, 2018-09-28:
In most cases those content packs come with a demo map which replicates the UE store images, but you also get the individual assets to create your own scene. As Triton stated, most royalty free licensing provides for binary distribution but no source distribution; make sure you read the various licensing models for purchased content.

eds89, 2018-09-28:
So, there's no way to change AirSim's source, which is available free of
charge, compile it again on an existing map (e.g. Neighbourhood)?

I'd like to add a camera effect, similar to Semantic Segmentation, which is
modelled through a PostProcessMaterial (I believe).

TritonSailor, 2018-09-28:
That's exactly what you do. Use any version of AirSim you want, and compile
it with your project that has your custom environment.

closed 2018-12-07 by madratman