microsoft/AirSim #1418: Linux AirLib update

Bill1235813, 2018-09-24:
My team is working on a data-intensity project with AirSim environment. We really want a AirLib for Linux so we can work with headless mode on clusters, but it seems no way to make it now. When will you update the Linux Binaries?

TritonSailor, 2018-09-24:
You can try this linux binary.  Built yesterday with latest AirSim master and UE4 4.20.2

https://www.dropbox.com/s/iuzh03fm76vinzq/Redtail.tgz?dl=0

It's a forest trails environment.  Runs sloooow compared to the Win64 version, but if it's in headless mode maybe won't matter, since the FPS is limited by the draw time.  I haven't tried headless.

Bill1235813, 2018-09-25:
@TritonSailor Thanks, that's cool. But could you explain how to build this linux binary? We have a different map so we need to build it by our own.

DavidLSmyth, 2018-09-25:
We're also hoping to build a linux binary for a custom map, any guidelines on how to do this would be fantastic

DavidLSmyth, 2018-09-25:
Thanks for the info, we'll give it a go and hopefully will be able to fix any errors. Thanks for the offer, we may take you up on it!

Bill1235813, 2018-09-29:
Thanks for the help of @TritonSailor, finally I made it with latest AirSim master and UE4 4.20.3. I followed the step of issue #1333 first. But I still had the problem of enabling PCH exceptions. I tried to disable PCH by ``bUsePCHFiles = false`` in *yourproject*/Source/*yourproject*.Target.cs. This fixed my PCH problem in Ubuntu 16.04, but I'm not sure if it'll work in 18.04.

If you have further error of ``FModuleManager``, you can add ``#include "Runtime/Core/Public/Modules/ModuleManager.h`` to Plugins/AirSim/Source/AirSimGameMode.cpp

If you still get the error of enabling PCH exceptions. Try running clean_rebuild.sh in AirSim.

closed 2019-02-19 by madratman