microsoft/AirSim #1421: Which are TOP 3 ways to scan real environment and transfer it to AirSim

Supaple-x, 2018-09-26:
Hey guys! 

I'm going to use my coper to scan environment landscape, rooms and so on into the .fbx model.

What choice do I have with this issue? Did you meet such cases? Which solutions are the best?

Many thanks

DavidLSmyth, 2018-09-26:
We currently are looking at two ways of doing this: 

- visualsfm to create the point cloud from images. Then import to blender or first import to meshlab, use meshlab to tidy the reconstruction up and then import to blender. Export from blender to fbx.

- OpenDroneMap to create the point cloud and convert the point cloud model to fbx using blender. 

I haven't looked into this much but both of the above options have yielded reasonable results. "Best" is a pretty subjective term, what are you looking to achieve exactly? There are metrics to compute the quality of a reconstruction but you obviously need the ground truth as a comparison

Supaple-x, 2018-09-26:
Thanks David,

I'll revise this ways....

I'm going to reconstruct real environment either natural objects such as trees, hills, holes, caves or bulitings and its internal decoration and architecture in high detalization.

Is it possible to do? 
Well, while a was writing this reply, I've revised visualfsm, and found out that it is out of development and support since 2012... May be it has some soft to be substituted?

DavidLSmyth, 2018-09-26:
Reconstruction of real environments is possible, if you search anything along the lines of "structure from motion" you'll come across a lot of useful information. I found Visual SFM suited my needs. If your criteria is that the software is being actively developed maybe have a look at opensfm.

If you're happy to pay, there are a number of commercial solutions such as pix4d and dronedeploy. Most of the softwares out there seem to use a similar pipeline, using SIFT/ORB for keypoint detection. An appropriate meshing algorithm will have a significant effect on the final product but to the best of my knowledge there isn't a huge difference with what goes on behind the scenes with opensfm and visualsfm, for example.

Your question is a very broad one, if you have some sample images I'd recommend using visualsfm to get an idea of how these types of software work (visualsfm is straightforward to use and will also take advantage of a gpu if you have one) and then refine your question.

RVBldr, 2018-09-26:
We've done some work on this as well using Pix4D for the SFM reconstruction. I was able to port directly from Pix4D to UE4, with mixed results. Depending on your visual requirements, the mesh port provides great reference material for an art team but you may need to adjust scale. I found that on import to UE, the source mesh was scaled much smaller than needed. Also, the number of source images, overlap, and point matching all have impacts on the mesh quality. 

I've attached two images, one, the screen shot from Pix4D with the SFM mesh reconstruction, and the 2nd, the import results into UE. 
[image]
[image]

closed 2018-12-07 by madratman