microsoft/AirSim #1423: How to load my own map by the Airsim

chengjinjack, 2018-09-27:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----

MaximStrnad, 2018-10-01:
You have to crate your own ureal engine project, then inject or create a map into it. Then you install the AirSim as plugin into the unreal engine and change the world game settings to AirSim game (something like that). Maybe you will need to delete all player "cameras" so if you start the game with only one player camera the airsim will automatically sign into it. There is a video on youtube explaining the full process, there is little to now comentary but if you slow it down you should be copy the moves. I hope that helps

NextSim, 2018-10-01:
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md

closed 2018-10-17 by sytelus