microsoft/AirSim #1431: Reverse car using Python SDK?

recumbentbirder, 2018-10-02:
1. in 'CarControls.set_throttle()', the references to 'self.'... are missing -- so nothing will be set using this method

2. Even though I set 'manual_gear' to -1 and 'throttle' to e.g. -0.2, the car doesn't actually go in reverse in the simulation (drives forward with `abs(throttle)` speed).

```
<CarControls> {   'brake': 0,
    'gear_immediate': True,
    'handbrake': False,
    'is_manual_gear': False,
    'manual_gear': -1,
    'steering': 0,
    'throttle': -0.2}
```

What is the correct setting of 'CarControls' to go into reverse?

sytelus, 2018-10-16:
See this example to [reverse car](https://github.com/Microsoft/AirSim/blob/a4fcf568e75b6d675d5d127754757d673ce33ecb/PythonClient/car/hello_car.py#L33).

closed 2018-12-07 by madratman

mpautzke, 2020-01-10:
Hopefully this helps someone.  I had the same issue, could not get the API to drive in reverse.  I am using Unity and found that user input set the footbrake = throttle, but the API did not (Assets/AirSimAssets/Scripts/Vehicles/Car #72).  Setting footbrake = throttle for the API block is one way to fix it or send in the same value for brake