microsoft/AirSim #1435: Camera/Image API doesn't work after update to version 1.2 + pip python client out of date warning

michael-fonder, 2018-10-04:
Hi,

I have a script which is perfectly working with airsim on the UE4.16, but for dependencies reasons, I need to move to UE4.18. So I'm trying to update my python script for using Airsim 1.2 in computer vision mode on my custom environment. I've followed the whole installation procedure and updated my code accordingly. However, I cannot get anything related to cameras to work properly (the rest of the api seems to work properly).

First, I faced the same error than the one highlighted in issue #1415. Then I tried to dig further with the hope of finding the problem. This only led to other problems. For example, a call of `simGetCameraInfo` showed me that parameters entered in the "CameraDefaults" field  of the settings didn't have any impact on the output of the function. So for now, I'm completely unable to customize/add any camera which is really frustrating since this feature is really mandatory for me.

Did anyone faced the same issue and/or could help me?

By the way, the call of `simGetCameraInfo` first led to the error `AttributeError: 'CameraInfo' object has no attribute 'proj_mat'`. This appeared to be due to the pip python client to be badly out of date. I indeed installed the client with pip three days ago and it appears that this error was corrected on the repository in July. It may be a good idea to plan an update of the package in a close future.

sytelus, 2018-10-17:
Could you please copy and paste settings.json that you were using? We have tested CameraDefaults and it works as expected. You can see [example code for simGetCameraInfo](https://github.com/Microsoft/AirSim/blob/a4fcf568e75b6d675d5d127754757d673ce33ecb/PythonClient/computer_vision/cv_mode.py#L21) which also works expected. The pip package is also up to date.

madratman, 2018-12-07:
@michael-fonder can you paste your settings.json. If this was resolved, please let us know?

madratman, 2018-12-19:
I'm closing coz of lack of response. please re-open and share your json file.

closed 2018-12-19 by madratman