microsoft/AirSim #1439: MAVLINKcom lib

jorisNas, 2018-10-05:
Hi,
I'm working on a project which aim to build a data exchange between a 3Dviewer and PX4. So I want to use the mavlink protocol. Then i found MavlinkCom cross platform library respond well to my goals with it classes and functionning process. But i still have some questions : 

1- Is it possible to make it work/use independantly of airsim?

2- working with VS2013 AND VS2015 , i want to know if there is a version or project file of MavlinkCom library that can easy being build in those Visual studio versions ?

3- And for the mavlinktest project a better tutorial/document on it differents functionning purpose ?

thanks ,JorisNAS

sytelus, 2018-10-16:
Yes, MavLinkCom can be used independently. Compiling on older VS shouldn't be problem although you will need to change vcproj files. The code for MavLinkTest is fairly straightforward.

madratman, 2018-12-07:
@jorisNas is this resolved?

lovettchris, 2018-12-18:
Yes mavlinkcom can be used independently, in fact, any mavlink toolkit will wok.  See how the LogViewer shows a 3D model of the current drone pose:
[image]

closed 2018-12-18 by madratman

ariel-elbit, 2020-09-13:
I'm trying to use the MavlinkCom project in order to get telemetries from mavlink connected via udp port using a sitl in the background. can't manage to figure it out. Anyone able to help?
Thanks a lot

kcee454, 2024-02-27:
Hi @ariel-elbit , have you solved this issue?