microsoft/AirSim #1441: Improvements ideas

nikonodein, 2018-10-05:
Hey all,

I wanted to discuss a couple of improvements I and my company would like to see in AirSim, and the viability of getting some of this changes implemented.

First of all my interest lie mainly in using AirSim purely for the graphics portion. Thus most of my improvements relate to using AirSim separately from the simulated physics and such. Also for most of these ideas I have implemented it in our Fork, so I would be happy to make pull requests for AirSim master.

1. Allow for tick commands for the steady clock via RPC.
This is mainly for running the simulation physics separately. This way you can ensure that the graphics and the simulation stay in sync. 

This is trivial to implement. Just add a function to rpclib lib to call the clock tick command. 

2. Create some kind of asynchronous publisher(especially for images)
Due to the request and response nature of the simGetImages() command, it was too slow for our purposes. When we ran our simulation their were noticeable lags every time we requested an image. Especially when we were requesting images the size of 1024x1024. Thus instead we copied the Api of the RecordingThread, and instead of recording to disk made it publish the images via ZMQ. This has worked reasonably well for our purposes, especially in conjunction with the tick command recommended above.

For the implementation as I mentioned something like the RecordingThread works very well. There would have to be some discussion on how we do the publishing. Either some basic sockets with TCP, or as we have done with ZMQ. The probably with the ZMQ is that we are adding another dependency, and I don't know how the community feels about this.

3. Query the server for all the meshes in the scene. 
Since on our side we are also doing path planning, we needed access to the underlying meshes. Thus I added a RequestMesh function to the RPC.  That returns a vector of MeshResponse which is similar to ImageResponse. Below is the struct that we use. 

To implement this functionality I have added the necessary commands and structs to the RPC server and client. I also worked out how to grab the Mesh vertices and triangles off of the GPU in packaged projects. This is a very slow function as it freezes the renderer. However, for now we only call this function once at the start. We also for now only deal with static meshes. Deformable meshes, and meshes that are not yet loaded in the scene still need to be discussed.

~~~~
struct MeshResponse {
	Vector3r position;
	Quaternionr orientation;

	std::vector<float> vertices;
	std::vector<uint32_t> indices;
	std::string name;
};
~~~~

As I mentioned above mainly these ideas are recommended in order to run the simulation and graphics separately. I also believe the slowdown is acceptable in certain places, as the functionality should be there more for power users that understand the extra costs certain functions may incur.

Nikolaus Mitchell
NodeIn Roboticist
nikolausmitchell@nodein.com

nikonodein, 2018-11-02:
Hey Shital,

Sorry for responding so late.

I have added a pull request for option 3.

For option 1).
The reasoning for the tick() command is to synchronize the clock between the offboard physics and the airsim. As far as I can tell the pause and continue API's for AirSim only deal with the running of the unreal engine game game loop. It does not allow for fine grained control of the SteadyClock. By controlling the steady clock I can control the rate at which the cameras record images.

closed 2018-11-02 by nikonodein

laurelkeys, 2020-10-12:
Hey @nikonodein did you happen to open-source your integration with ZeroMQ? I was starting to look into it, for reasons similar to those you mentioned (passing large images, with real-time constraints). Thanks!