microsoft/AirSim #1446: How to build my own road in AirSim or UE4 for training cars?

HolyBugger, 2018-10-08:
When I am using AirSim to train Car, I need to build some special environment. For example ,including peoples, roads, buildings. How can i do that ? Is it possible that importing an opendrive file or OSM file to AirSim?

diarmaidocualain, 2018-10-08:
I dont know about opendrive or OSM, but you can create your own AirSim environment using the Unreal Editor. There is a guide available here for importing the AirSim plugin into another environment here, you would just have to create your own first : https://www.youtube.com/watch?v=1oY8Qu5maQQ

GuangTongLiu1, 2018-10-09:
Ok. Thank. Opendrive and OSM is a format of map that containing environment messages （e.g. road, building). I dont know how to build a road in UE4. But I have to do that for training my car in AirSim. Can you help me？

TritonSailor, 2018-10-09:
This is a very useful feature. If performance needs to be optimized, it would probably be helpful to ask in the unreal forums how to most efficiently implement this.  Also, you might find some useful code in the Nvidia Waveworks project.  It uses ray casting (on the gpu) to determine bouyancy of actors at various locations on the map.

diarmaidocualain, 2018-10-09:
Have you looked at the City environments that AirSim comes pre-built with here to see if they are suitable? 
Or else have a look at the roads and buildings in the Unreal Marketplace, there are many there and could be suitable for your needs, or else a good starting point to modify for your needs.

GuangTongLiu1, 2018-10-10:
Ok. I find some tutorials for building road by spline tool in UE4. But I am still learning how to use it. The links are as follows:
 https://docs.unrealengine.com/latest/INT/Engine/Landscape/Editing/Splines/index.html

https://docs.unrealengine.com/latest/INT/Resources/ContentExamples/Landscapes/1_2/index.html

https://www.youtube.com/watch?v=JoCPi-9Op5U

https://youtu.be/H6xhXxRWwAI

closed 2018-11-01 by HolyBugger