microsoft/AirSim #1451: Manipulating characterObject (PhysX car) from within UE4

baaaze, 2018-10-10:
Hey,

I've been looking around for a while and I can't seem to find how I can call the character object (The PhysX car). I want to be able to manipulate it to follow a spline when ordered, in order to fake Autonomous driving. Appreciate all the insights I can get!

Thank you!

hesham-akmal, 2018-10-10:
Have you tried opening "AirSimAssets" in the "Content Broswer" and selecting "Edit SuvCarPawn" in the "World Outliner"

baaaze, 2018-10-11:
> Have you tried opening "AirSimAssets" from the "Content Browser" and selecting "Edit SuvCarPawn" from the "World Outliner"

That would be awesome! :-) But I can't even see that. All I have is Assets and Maps from the standard Landscape Mountain stuff. Do I need to manually import the Airsim asset aside from loading the plugin?

hesham-akmal, 2018-10-11:
oh I assumed you're on the Blocks level. Okay try this, in the "View Options" in the Content Browser, there's an option to show plugin content, check it, you'll be able to navigate to the plugin content.

baaaze, 2018-10-11:
> oh I assumed you're on the Blocks level. Okay try this, in the "View Options" in the Content Browser, there's an option to show plugin content, check it, you'll be able to navigate to the plugin content.

Dude! You are awesome! Thank you for saving me from tons of headaches!

hesham-akmal, 2018-10-11:
Anytime! Good luck with your project

sytelus, 2018-10-16:
If you want a fake car to follow spine, you can just do some bluescripting in UE on a car mesh. You can't control PhysX based car to follow spine precisely.

closed 2019-02-19 by madratman