microsoft/AirSim #1456: Relationship between camera_position, kinematics_estimated.position and position in airsim_rec.txt, and

diarmaidocualain, 2018-10-11:
Hi! 
I have been reading the documentation and playing around with Airsim and I am very impressed with what I have seen so far. 
I have made a simple test with the Python API where I get a multirotor drone to follow a short path and then land. The position data is simultaneously logged to airsim_rec.txt by the plugin. 

```
TimeStamp	POS_X	POS_Y	POS_Z	Q_W	Q_X	Q_Y	Q_Z	ImageFile
1539453811633	0.000000	0.000000	-0.141340	0.694659	0.000000	0.000000	-0.719339	img__0_1539184758841717600.png
1539453812389	0.000000	0.000000	-0.141340	0.694659	0.000000	0.000000	-0.719339	img__0_1539184759031287600.png
```

This positon data is shown by the blue crosses in the Matlab plot below. After landing, we wait for 1 second and then a set of stereo images are requested along with the camera_position of each camera and written to file. 

```
responses = client.simGetImages([
    airsim.ImageRequest("front_left", airsim.ImageType.Scene), #scene vision image in png format
    airsim.ImageRequest("front_right", airsim.ImageType.Scene)]) #scene vision image in png format

camGroundTruth = open((outDir + 'camGroundTruth.txt'), "w")
camGroundTruth.write("imgName timestamp cameraPositionX cameraPositionX cameraPositionZ cameraOrientationW cameraOrientationX cameraOrientationY cameraOrientationZ\n")

for response in responses:
    filename = (outDir + str(idx))
    camGroundTruth.write(" %s %f %f %f %f %f %f %f %f\n" %
                  (filename, response.time_stamp, response.camera_position.x_val, 
                   response.camera_position.y_val, response.camera_position.z_val,
                   response.camera_orientation.w_val, response.camera_orientation.x_val,
                   response.camera_orientation.y_val, response.camera_orientation.z_val))

```

The camera_position pose of each camera is plotted in Matlab, as shown by the pair of green, red and blue axis below. The drone is queried for its kinematics_estimated.position after landing too. 

```
multirotorState = client.getMultirotorState()
multirotorStateKinematicsEstimatedPosition = multirotorState.kinematics_estimated.position
multirotorStateKinematicsEstimatedOrientation = multirotorState.kinematics_estimated.orientation
```

which is shown by the third axis below. 

[image]

From this test, we would expect that the last drone center point location recorded in airsim_rec.txt is the "ground truth" of where the drone captured the images from after landing. We would also expect that the camera_position's are "ground truth" locations too, and as a result, are located 25cm from each other on the drone and at an equal distance from the drone centerpoint. 

Calculating the distances, I have found that the cameras are indeed exactly 25cm from each other. However, the left and right cameras are 0.6196m and 0.6234m from the centre point of the drone respectively. Would anyone have a suggestion as to why this is the case? 

Also worth noting is the calculated distance of the camera locations to the estimated `kinematics_estimated.position` point is 0.476680950055695m and 0.476680993257545m for the left and right camera respectively. It appears that the estimated position is located at a point incredibly accurately equidistant to both cameras. Is the estimated position expected to be that accurate? 

Thanks!

sytelus, 2018-10-17:
Welcome @diarmaidocualain to AirSim community! The reason cameras are off the center is because they are mounted in the front of vehicle. I think the vehicle is 1X1 meter.

Currently we are passing ground truth for estimated kinematics when using SimpleFlight. After we implement state estimation in SimpleFlight, we will be able to use actual estimated state and in that case things won't be so inaccurate.

bimalkmehta, 2018-12-07:
Been a while since no response

closed 2018-12-07 by bimalkmehta