microsoft/AirSim #1461: Blocks Build Error building failes

juanjuanfire, 2018-10-12:
Whille trying to open block i get an error saying that libUE4Editor-Blocks.so is missing and so is the libUE4Editor-AirSim.so 

Here is my console log
Running Mono...

~/UnrealEngine/Engine ~/UnrealEngine
Using 'git status' to determine working set for adaptive non-unity build.
@progress push 5%
@progress 'Generating code...' 0%
@progress 'Generating code...' 67%
@progress 'Generating code...' 100%
@progress pop
Performing 4 actions (4 in parallel)
@progress 'Compiling C++ source code...' 0%
[2/4] Compile Module.AirSim.cpp
[1/4] Compile Module.AirSim.gen.cpp
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:6:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/CameraDirector.gen.cpp:8:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/PawnSimApi.h:18:
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/VehicleApiBase.hpp:89:29: error: 
      default initialization of an object of const type 'const
      msr::airlib::RCData' without a user-provided default constructor
        static const RCData invalid_rc_data;
                            ^
                                           {}
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:4:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/src/api/RpcLibServerBase.cpp:10:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/RpcLibServerBase.hpp:9:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/ApiProvider.hpp:7:
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/VehicleApiBase.hpp:89:29: error: 
      default initialization of an object of const type 'const
      msr::airlib::RCData' without a user-provided default constructor
        static const RCData invalid_rc_data;
                            ^
                                           {}
1 error generated.
@progress 'Compiling C++ source code...' 20%
@progress 'Compiling C++ source code...' 40%
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:20:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:12:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:7:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:188:46: error: 
      conditional expression is ambiguous; 'const Quaternion<[...], 2>' can
      be converted to 'Quaternion<[...], (default) AutoAlign aka 0>' and
      vice versa
        QuaternionT ref_n = assume_unit_quat ? ref : ref.normalized();
                                             ^ ~~~   ~~~~~~~~~~~~~~~~
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp:110:52: note: 
      in instantiation of member function
      'msr::airlib::VectorMathT<Eigen::Matrix<float, 3, 1, 0, 3, 1>,
      Eigen::Quaternion<float, 2>, float>::rotateQuaternion' requested here
    msr::airlib::Quaternionr ray_q_b = VectorMath::rotateQuaternion(ray_...
                                                   ^
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:20:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:11:
In file included from /home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Settings.hpp:7:
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils//Utils.hpp:138:20: error: 
      returning reference to local temporary object
      [-Werror,-Wreturn-stack-address]
            return default_val;
                   ^~~~~~~~~~~
/home/juan/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/WorldSimApi.cpp:204:36: note: 
      in instantiation of function template specialization
      'common_utils::Utils::findOrDefault<map, std::__1::basic_string<char>,
[... 55 lines cut ...]
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/PhysXVehicles/Binaries/Linux/libUE4Editor-PhysXVehiclesEditor.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/CableComponent/Binaries/Linux/libUE4Editor-CableComponent.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/MobilePatchingUtils/Binaries/Linux/libUE4Editor-MobilePatchingUtils.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/ExampleDeviceProfileSelector/Binaries/Linux/libUE4Editor-ExampleDeviceProfileSelector.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/CustomMeshComponent/Binaries/Linux/libUE4Editor-CustomMeshComponent.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/AndroidPermission/Binaries/Linux/libUE4Editor-AndroidPermission.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/ArchVisCharacter/Binaries/Linux/libUE4Editor-ArchVisCharacter.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/Steam/SteamVR/Binaries/Linux/libUE4Editor-SteamVR.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/Steam/SteamVR/Binaries/Linux/libUE4Editor-SteamVRController.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/ImmediatePhysics/Binaries/Linux/libUE4Editor-ImmediatePhysics.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/ImmediatePhysics/Binaries/Linux/libUE4Editor-ImmediatePhysicsEditor.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/LocationServicesBPLibrary/Binaries/Linux/libUE4Editor-LocationServicesBPLibrary.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Messaging/TcpMessaging/Binaries/Linux/libUE4Editor-TcpMessaging.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Messaging/UdpMessaging/Binaries/Linux/libUE4Editor-UdpMessaging.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystem/Binaries/Linux/libUE4Editor-OnlineSubsystem.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystemNull/Binaries/Linux/libUE4Editor-OnlineSubsystemNull.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystemUtils/Binaries/Linux/libUE4Editor-OnlineSubsystemUtils.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystemUtils/Binaries/Linux/libUE4Editor-OnlineBlueprintSupport.so (API version 0).
[... cut ...]
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/2D/Paper2D/Binaries/Linux/libUE4Editor-PaperSpriteSheetImporter.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/2D/Paper2D/Binaries/Linux/libUE4Editor-PaperTiledImporter.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/2D/Paper2D/Binaries/Linux/libUE4Editor-SmartSnapping.so (API version 0).
LogInit: Warning: Still incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Still incompatible or missing module: libUE4Editor-AirSim.so

deksprime, 2018-10-13:
Having the same problem while building

```
In file included from /home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:20:
In file included from /home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:12:
In file included from /home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:7:
In file included from /home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:188:46: error: 
      conditional expression is ambiguous; 'const Quaternion<[...], 2>' can
      be converted to 'Quaternion<[...], (default) AutoAlign aka 0>' and
      vice versa
        QuaternionT ref_n = assume_unit_quat ? ref : ref.normalized();
                                             ^ ~~~   ~~~~~~~~~~~~~~~~
/home/x/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp:110:52: note: 
      in instantiation of member function
      'msr::airlib::VectorMathT<Eigen::Matrix<float, 3, 1, 0, 3, 1>,
      Eigen::Quaternion<float, 2>, float>::rotateQuaternion' requested here
    msr::airlib::Quaternionr ray_q_b = VectorMath::rotateQuaternion(ray_...
                                                   ^
1 error generated.
```

Anyone?

ChanganVR, 2018-10-14:
I met the same problem...

BrunoXL, 2018-10-15:
Me too, did anyone found a solution?

ChanganVR, 2018-10-15:
Rewriting the code as below could temporarily fix the issue.
`        QuaternionT ref_n, ref_n_i;
    	if (assume_unit_quat) {
	    ref_n = ref;
            ref_n_i =  ref.conjugate();
		}
    	else{
            ref_n = ref.normalized();
            ref_n_i =  ref.inverse();
        }
//        QuaternionT ref_n = assume_unit_quat ? ref : ref.normalized();
//        QuaternionT ref_n_i = assume_unit_quat ? ref.conjugate() : ref.inverse();`

ivannson, 2018-10-16:
@ChanganVR which file is this code in?

ChanganVR, 2018-10-16:
@ivannson The file is AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp

sytelus, 2018-10-16:
This change is now merged to master.

closed 2018-10-16 by sytelus