microsoft/AirSim #1465: lidar build error

hdy117, 2018-10-15:
I have enabled lidar sensor according to lidar.md.
But got this error, when I rebuild AirSim.

STUBBED: FDesktopPlatformLinux::GetNativeFeedbackContext at /home/hdy/UnrealEngine_4.18/Engine/Source/Developer/DesktopPlatform/Private/Linux/DesktopPlatformLinux.cpp:269 (GetNativeFeedbackContext)
Running /home/hdy/UnrealEngine_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE

Running Mono...

~/UnrealEngine_4.18/Engine ~/UnrealEngine_4.18
Using 'git status' to determine working set for adaptive non-unity build.
@progress push 5%
@progress 'Generating code...' 0%
@progress 'Generating code...' 67%
@progress 'Generating code...' 100%
@progress pop
Performing 3 actions (8 in parallel)
@progress 'Compiling C++ source code...' 0%
[1/3] Compile Module.AirSim.cpp
In file included from /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
In file included from ../../../AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirBlueprintLib.h:20:
In file included from /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:12:
In file included from /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirLib/include/common/CommonStructs.hpp:7:
In file included from /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:188:46: error: 
      conditional expression is ambiguous; 'const Quaternion<[...], 2>' can be converted to
      'Quaternion<[...], (default) AutoAlign aka 0>' and vice versa
        QuaternionT ref_n = assume_unit_quat ? ref : ref.normalized();
                                             ^ ~~~   ~~~~~~~~~~~~~~~~
/home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp:110:52: note: 
      in instantiation of member function 'msr::airlib::VectorMathT<Eigen::Matrix<float, 3, 1, 0, 3,
      1>, Eigen::Quaternion<float, 2>, float>::rotateQuaternion' requested here
    msr::airlib::Quaternionr ray_q_b = VectorMath::rotateQuaternion(ray_q_l, lidar_pose.orien...
                                                   ^
1 error generated.
@progress 'Compiling C++ source code...' 25%
@progress 'Compiling C++ source code...' 50%
ERROR: UBT ERROR: Failed to produce item: /home/hdy/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

hdy117, 2018-10-15:
I am not very familiar with c++. I change the code to:

if(assume_unit_quat){ 
        	ref_n=ref;
        }
        else{
        	ref_n=ref.normalized();
     	}
then it gives me a pass

BrunoXL, 2018-10-15:
This issue mensions same error from #1461 
The original code is `QuaternionT ref_n = assume_unit_quat ? ref : ref.normalized();`
He just rewrite it.
I'm wondering with whether it is related to compile, once nobody posted a issue about that on Windows

bmalhigh, 2018-10-16:
I created a pull-request for this issue. Let me know if those changes fix the overall issues on Ubuntu.
https://github.com/Microsoft/AirSim/pull/1470

sytelus, 2018-10-16:
This change is merged now in master.

closed 2018-10-16 by sytelus