microsoft/AirSim #1466: Failed to build Blocks [Mac]

ivannson, 2018-10-16:
Installed UE 4.18.3, git pulled from AirSim repo following instructions for Linux. When I try opening the Blocks project following the docs (selecting in-place conversion), I get the following error message:
ERROR: UBT ERROR: Failed to produce item: /Users/novikov/AirSim/Unreal/Environments/Blocks 

I'm running macOS Mojave and Xcode 9.0 (same thing happened when I was using Xcode 10.0).

Full log:

Running Mono...

Setting up Mono
~/UE_4.18/Engine ~/UE_4.18/Engine/Binaries/Mac
@progress push 5%
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/Users/novikov/AirSim/Unreal/Environments/Blocks 4.18/Blocks.uproject" "/Users/novikov/AirSim/Unreal/Environments/Blocks 4.18/Intermediate/Build/Mac/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Reflection code generated for BlocksEditor in 17.182893 seconds
@progress pop
Performing 6 actions (4 in parallel)
[1/6] Compile SharedPCH.Engine.h
error: unknown warning option '-Wno-unused-lambda-capture'; did you mean '-Wno-unused-parameter'? [-Werror,-Wunknown-warning-option]
ERROR: UBT ERROR: Failed to produce item: /Users/novikov/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib
Total build time: 38.38 seconds (Local executor: 0.00 seconds)

sytelus, 2018-10-16:
I think this is issue with UE 4.18. See here: https://answers.unrealengine.com/questions/779733/xcode-build-errors-lambda-capturenot-used.html

madratman, 2018-12-07:
@ivannson was this resolved for you?

ivannson, 2018-12-10:
@madratman No, I gave up on trying to install on macOS and used Windows through Bootcamp instead