microsoft/AirSim #1467: How to modify the position of the camera

liyisecond339, 2018-10-16:
Hi:
I used the recording feature to get all of the camera image of the vehicle. the environment i used is 
"Warehouse", below is my config in setting.json
{
  "SettingsVersion": 1.2,
  "SimMode": "",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraName": "front_center",
        "Compress": true,
        "ImageType": 0,
        "PixelsAsFloat": false
      },
      {
        "CameraName": "front_right",
        "Compress": true,
        "ImageType": 0,
        "PixelsAsFloat": false
      },
      {
        "CameraName": "front_left",
        "Compress": true,
        "ImageType": 0,
        "PixelsAsFloat": false
      },
      {
        "CameraName": "fpv",
        "Compress": true,
        "ImageType": 0,
        "PixelsAsFloat": false
      },
      {
        "CameraName": "back_center",
        "Compress": true,
        "ImageType": 0,
        "PixelsAsFloat": false
      }
    ]
  }
}
The picture i got like below:
front_left:
[image]
front_right:
[image]

my question is, from the image, the left and right image is quite same, it is possible to change the cameras directions?

another question is, it is possible to get the frame at the same time? the time stamp is nano seconds？

sytelus, 2018-10-16:
Looks like you are old version of settings.json schema. In AirSim V1.2, the "Cameras" section is moved under "Vehicles".

liyisecond339, 2018-10-17:
thanks a lot!
you mean is like below?
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      **"Cameras": {   
        //same elements as CameraDefaults above, key as name
      },**
}
as my understanding the define will cover the CameraDefaults settings. but in the CameraDefualts element i did not find the where to define the camera`s direction.

RVBldr, 2018-12-07:
@liyisecond339 Is this issue resolved?

orangelx, 2020-03-27:
"Vehicles": {
    "Drone1": {
      "VehicleType": "SimpleFlight",
      "Cameras": {
        "MyCamera1": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 640,
              "Height": 320,
              "FOV_Degrees": 100
            }
          ],
          "X": -0.1,
          "Y": 0,
          "Z": -0.4,
          "Pitch": 0,
          "Roll": 0,
          "Yaw": 0
        }
}
      "X": 0,
      "Y": 0,
      "Z": 0,
      "Pitch": 0,
      "Roll": 0,
      "Yaw": 0
}
}

jonyMarino, 2020-08-10:
@orangelx did you reproduce the issue with that values?

closed 2022-03-31 by jonyMarino