microsoft/AirSim #1471: Reproduciability of AirSim-trained networks in real autonomous driving situations

eds89, 2018-10-16:
Hello guys,

Have you ever seen an assessment of how well neural networks trained in MS AirSim's simulated world work in real autonomous driving situations? I'd love to see a detailed scientific study on this. Some top researchers at my university disregard simulation technology altogether.

DavidLSmyth, 2018-10-16:
In relation to AirSim, I saw these at IJCAI this year:
https://www.ijcai.org/proceedings/2018/0678.pdf
https://www.ijcai.org/proceedings/2018/0847.pdf

It seems to me like there's plenty of interest in using simulations and transfer learning to train systems due to the fact that simulations offer a ground truth, but I don't know of any detailed studies outlining a methodology evaluating their effectiveness. I'd also be interested if anyone has any quantitative results or methodology.

ashkapoor, 2018-12-07:
See: https://github.com/Microsoft/AirSim/wiki/technion

closed 2018-12-07 by madratman