microsoft/AirSim #1472: Issue when recording segmentation images

Nozomeister, 2018-10-17:
Hi AirSim team,
Thank you for providing this awesome simulator.
There is a problem that I encountered which is when I tried to use the record function, I could save the RGB image with ImageType = 0 in setting.json with the resolution of 1920x1080, but when I tried to save the image in segmentation mode (ImageType = 5), the simulator did not save the images and the out put were 256 x 144 RGB images.
Could you help me with it? 
Thank you.

sytelus, 2018-10-17:
Could you please post your settings.json?

Nozomeister, 2018-10-17:
[settings.json.tar.gz](https://github.com/Microsoft/AirSim/files/2485718/settings.json.tar.gz)
Here is my settings file.

Nozomeister, 2018-10-23:
Hi @sytelus, is there any updates regarding this issue? This has been troubling me for a while now.

vaschindler, 2018-10-25:
In your settings file you just specified ImageType 4 in CameraDefaults, but you need to specify every image type you want to record. Something like this: 

"CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1920,
        "Height": 1080,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
	   {
        "ImageType": 5,
        "Width": 1920,
        "Height": 1080
     
      }, ...
]
}

RVBldr, 2018-12-07:
@Nozomeister Is this resolved?

yangautumn, 2019-04-06:
The solution works for me, thanks!

closed 2019-04-07 by madratman