microsoft/AirSim #1483: How can I change the content of the Subwindow?

WadeWuu, 2018-10-22:
I know that the content of subwindow  is from the camera on the vehicle. However, now I want it to display another thing. For example, I want it to display my own images.
This work is a first step. My lab wants to add a new sensor which is not included on AirSim to capture the images and use the subwindow to display. Now, I want to know how the subwindow works and how to change it content.

DavidLSmyth, 2018-10-22:
Subwindows are initialised here, you could hard code in changes:
https://github.com/Microsoft/AirSim/blob/5ad5b98bb9233f492c7a78553158f77cdaa1cd4c/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L308-L341

WadeWuu, 2018-10-27:
After I change it, should I compile the plugins?

WadeWuu, 2018-10-27:
After I change it, should I compile the plugins? or just restart the project of UE.

WadeWuu, 2018-10-28:
Yeah, I have read the code.  The variable subwindow_setting  records the information of the subwindow and is used to set up the subwindow_cameras_. I think subwindow_cameras_ is used to capture the images and display them on the Unreal Engine. But I do not know how subwindow_cameras_ works. I know subwindow_cameras_ is from the class APIPCamera, but APIPCamera is a little complicate.  I can not understand.
Can you explain how subwindow_cameras_  capture  the images and display them on Unreal Engine?
Thank you.

WadeWuu, 2018-10-30:
Yeah, I have read the code. The variable subwindow_setting records the information of the subwindow and is used to set up the subwindow_cameras_. I think subwindow_cameras_ is used to capture the images and display them on the Unreal Engine. But I do not know how subwindow_cameras_ works. I know subwindow_cameras_ is from the class APIPCamera, but APIPCamera is a little complicate. I can not understand.
Can you explain how subwindow_cameras_ capture the images and display them on Unreal Engine?
Thank you.  @DavidLSmyth

zhuboshun2015, 2019-05-30:
@DavidLSmyth Do you know how to set the subwindows for multiple vehicles ?

DavidLSmyth, 2019-05-30:
> @DavidLSmyth Do you know how to set the subwindows for multiple vehicles ?

Assuming you mean you want a subwindow showing the scene view for each vehicle, I'd advise having a look through https://github.com/saihv/AirSimMulti/, which is now deprecated but outlines the more basic changes that were made to AirSim to facilitate multiple vehicles. Specifically, lines 43-52 of this file:
 https://github.com/saihv/AirSimMulti/blob/cfc1fc7eb2f7b6e67c7cb59047e9b3273da4691b/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp

You'll have to make a couple of changes but it shouldn't be too hard. I'm sorry but I don't have the time to outline them fully.

closed 2021-04-30 by jonyMarino