microsoft/AirSim #1486: Is it possible to add custom objects to the environment such as traffic cones?

bsridatta, 2018-10-23:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----

TritonSailor, 2018-10-23:
Yes, in the editor, if you get the downtown environment it already has traffic cones.  You can move them wherever you want in the editor, or spawn them with blueprints, or use the traffic cone meshes in another environment.  

You can also change the collision to physics actors and  traffic cones will bounce properly off the vehicle if you adjust the physical material settings.

bsridatta, 2018-10-26:
Awesome!  thanks :)

closed 2018-10-26 by bsridatta