microsoft/AirSim #1490: what should i do if i want to use API to control my car?

GuangTongLiu1, 2018-10-24:
Hi, I am not familiar with Airsim. But I want to use API to control my car. I have seen the example which using python API to control car. But I still have some questions.
1. Where should i place code files ?
2. How can I get certain car's states if there are many cars in the project ?
3. What should I do if i want to build my own car models?

Hope you can help me ! Thanks very much!

bmalhigh, 2018-10-24:
w.r.t. #1 -- I am not sure I understand the question -- but you will place the code in your own Python/C++ client code that communicates with AirSim.
w.r.t #2 -- The getter on the state APIs allow you to pass an optional vehicle-name, e.g.,
                client.getMultirotorState(vehicle_name="drone1")
                See multi_agent_drone.py for more details.

w.r.t. #3 -- I am not sure what you mean by a "model" here. Can you elaborate more on what aspects you want to change for your own car model?

GuangTongLiu1, 2018-10-25:
> w.r.t. #1 -- I am not sure I understand the question -- but you will place the code in your own Python/C++ client code that communicates with AirSim.

1. Ok. Thanks for your answers . Maybe i should elaborate more details. If i have the code files which can control the car, e.g.,the demo "AirSim/PythonClient/car/". Where should i place the files in ? Is the path "/(My UE4 project )/Plugins/Airsim"?  And then compile the whole project after placing the code files correctly? I am very  doubtful about that.

> w.r.t. #3 -- I am not sure what you mean by a "model" here. Can you elaborate more on what aspects you want to change for your own car model?
 
My own car model means that I model my car with respect to it's own kinetic and  dynamic characteristic . You know that there are obviously difference between different cars. So if I have my own car model ,for example , i have a simulink model of my car . Are there ways to use it ? Such as building my simulink model to C++ format files? Or is it possible to communicate with Matlab/simulink  when Airsim and simulink both are running?  

Hope your reply. Thanks a lot.

bmalhigh, 2018-10-25:
w.r.t. #1 -- you should be able to place the files any where as long as you have the airsim python module available.

w.r.t. #3 -- @sytelus -- would love your input on this.

closed 2018-11-08 by dahewujiang