microsoft/AirSim #1491: car_lidar.py Question

qazxsw1234, 2018-10-24:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----
Hello,
I execute car_lidar.py and I got a following error when run car_lidar.py.
what should I do if I want to use getLidarData() function?

My error message is as follows.
msgpackrpc.error.RPCError: rpclib: server could not find function 'getLidarData' with argument count 2.

pgashwin, 2018-10-24:
Did you try the latest PythonClient folder and updated Airsim?

qazxsw1234, 2018-10-24:
I just updated Airsim and tried the latest drone_lidar.py, but it doesn`t work.
Could you give me another suggestion?

bmalhigh, 2018-10-24:
@qazxsw1234 - can you clarify what error you are getting now after updating the airsim module?
Your first error message seems like a general case of the method not found.

bmalhigh, 2018-10-29:
This is an issue only for cars?

GuangTongLiu1, 2018-10-30:
Hi, @bmalhigh 
I tried the drone demo. It also has the same problem.
Is any mistake in my "settings.json" file?
The code is as follows:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "Vehicles": {

    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "AutoCreate": true,
      "Sensors": {
        "MyLidar1": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      }
    }
  }
}
`
I think Lidar1 doesn't work because there is no point cloud display in scenario even though i set the " "DrawDebugPoints": true ". How to resolve that problem?

Another question is that i found the "settings.json" file is shared with other projects in UE4? If some projects have different configuration, it is possible to link certain settings.json file to any UE4 project ?

Thanks!
Best regards!

bmalhigh, 2018-10-30:
@GuangTongLiu1 -- The first thing to check would be if you have the lastest AirSim plugin bits as part of your UE project. Remember you will need to update the copy of AirSim plug-in bits that are part of your UE project.

The AirSim settings.json is currently shared by all UE projects for a user.

GuangTongLiu1, 2018-10-31:
@bmalhigh  I am so careless that i forget to replace the old Airsim plug-ins of my UE4 projects with the new one. Now I have fixed the problem. Thanks !

closed 2018-10-31 by bmalhigh

PygmalionChen, 2019-03-13:
> @bmalhigh I am so careless that i forget to replace the old Airsim plug-ins of my UE4 projects with the new one. Now I have fixed the problem. Thanks !

I'm confused to replace the old Airsim plugins of my UE4 projects. My env is ubuntu, I run the released AirSimNH.sh and the car_lidar script to start. But where is the Airsim plugins? Could you explain it in details? Thanks!

vishal-prgmr, 2019-04-18:
@bmalhigh How do get the LIDAR APIs to work with the prebuilt binary that i downloaded? I am not using the source build.

Thank You