microsoft/AirSim #1493: Airsim Build Error

abhimanyuchadha96, 2018-10-25:
I am following the instructions as per this https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md

In the third step after I clone the AirSim and run the ./setup.sh it runs fine. 
But on running the ./build command, it gives me this error:-

putting build in build_debug folder, to clean, just delete the directory...
[  1%] Building CXX object rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o
clang: error: no such file or directory: 'Motion'
clang: error: no such file or directory: 'Planning/Project/UnrealEngine/AirSim/llvm-build/output/include/c++/v1'
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:62: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all' failed
make[1]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Could someone please help me resolve this?

yujianyuanhaha, 2018-10-25:
1. there should be folder like "llvm-build" after you finish the build, likely fail on build.
2. please launch Xcode to double check all essential plugins are installed
3. instead of build the Unreal Engine from codes, I start by install it of version4.20 from epic games launcher, not sure it matters

yujianyuanhaha, 2018-10-25:
quite some parsers are stupid at processing strings with space, like they fail at "Motion Planning" while can see "Motion_Planning",  take a try by renaming it.

madratman, 2018-12-07:
@abhimanyuchadha96 is this resolved now? If so, please close the issue.

closed 2018-12-10 by abhimanyuchadha96