microsoft/AirSim #1494: Captured images are not in sync with position and orientation

alexart13, 2018-10-25:
I recorded a few tracks in CV (FPV) mode and noticed an odd thing in the recordings: it seems that images are captured at one point of time and then position and orientation coordinates are captured a moment later. 

I tried to reproject the key points in the two consecutive images based on the coordinates recorded in airsim_rec.txt and got the following results:
[image]
[image]

As one can see, the key points shifted, but not to the expected position.
I looked into the code. In `UnrealImageCapture::getSceneCaptureImage` there are the following lines:

```
RenderRequest render_request(use_safe_method);
    render_request.getScreenshot(render_params.data(), render_results, render_params.size());

    for (unsigned int i = 0; i < requests.size(); ++i) {
        const ImageRequest& request = requests.at(i);
        ImageResponse& response = responses.at(i);
        APIPCamera* camera = cameras_->at(request.camera_name);
              
        response.camera_name = request.camera_name;
        response.time_stamp = render_results[i]->time_stamp;
        response.image_data_uint8 = std::vector<uint8_t>(render_results[i]->image_data_uint8.GetData(), render_results[i]->image_data_uint8.GetData() + render_results[i]->image_data_uint8.Num());
        response.image_data_float = std::vector<float>(render_results[i]->image_data_float.GetData(), render_results[i]->image_data_float.GetData() + render_results[i]->image_data_float.Num());

        msr::airlib::Pose pose = camera->getPose();
        response.camera_position = pose.position;
        response.camera_orientation = pose.orientation;
        response.pixels_as_float = request.pixels_as_float;
        response.compress = request.compress;
        response.width = render_results[i]->width;
        response.height = render_results[i]->height;
        response.image_type = request.image_type;
    }
```
There is a noticeable time gap between image capturing and coordinates capturing especially on not very fast computers. During this time gap the camera keeps moving. As a result the captured coordinates don't match with the real coordinates in which the image was captured.

As a quick workaround I added UGameplayStatics::SetGamePaused() to pause the physics just before image capturing and then unpause it after the image and the coordinates are captured. But I'm looking for a better solution.
The workaround helped as one can see below.

[image]
[image]
  
Do you have any suggestions how to get the coordinates at exactly the same moment as image capturing?

diarmaidocualain, 2018-10-31:
HI, thank you for this information. Just to add to your workaround, you can also pause the simulator from the python client code.     
# pause physics engine on drone while we capture data
    client.simPause(True)
.... 
# unpause physics engine on drone after we capture data
    client.simPause(False)

alexart13, 2018-10-31:
> As an aside, OP, what Unreal environment are you running? -I am looking for some decent indoor ones myself and I havent come across the one in your images before.

We use that one: https://www.unrealengine.com/marketplace/apartment-interior

yanenkoa, 2018-11-08:
#1505 didn't fix everything, unfortunately. Sim pawn coordinates are still captured _after_ the screenshot is taken, and on slow systems this difference can be quite significant, and it is visible even when running AirSim on high-end computers.
Pawn coordinates are captured in `RecordingFile::appendRecord` at [line 51](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Recording/RecordingFile.cpp#L51). And if you take a look at [lines 90-91](https://github.com/Microsoft/AirSim/blob/abda35ae403ce607ffd6ee610594ffb96b7064ad/Unreal/Plugins/AirSim/Source/Recording/RecordingThread.cpp#L90-L91) in `FRecordingThread::Run`, you can see that `RecordingFile::appendRecord` is only called after `UnrealImageCapture::getImages`, which captures screenshots, and in this gap the simulation keeps on running.
Thankfully, you can use camera coordinates to get accurate positions, which are captured in sync with the screenshot by calling `UnrealImageCapture::getImages`, but it would be nice to fix the described issue.

Amigoshan, 2019-04-14:
> HI, thank you for this information. Just to add to your workaround, you can also pause the simulator from the python client code.
>
> ```
> # pause physics engine on drone while we capture data [...]

I'm also bothered by the synchronization problem for a long time, until I found the walkaround suggested by alexart13. However, the python interface simPause is not doing the same thing from my understanding. 

It only pause the physical engine instead of the unreal rendering. So if there are moving objects in the environment, they would still look different in different images. 

What I did is following alexart13's way, in the WorldSimApi.cpp file, modify the pause function as follows: 

`void WorldSimApi::pause(bool is_paused)
{
    UGameplayStatics::SetGamePaused(simmode_->GetWorld(), is_paused);
    simmode_->pause(is_paused);
}`

In this way, the whole world will freeze whenever the pause function is called. The only issue is that it returns black images at the first time I query the images by simGetImages(). But everything seems OK after the first query.

jonyMarino, 2020-08-14:
@Amigoshan the PR solved this issue?

PR #3483 (2021-03-23): Add quick tip for syncing image API with physics state

Fixes: #1494

## About
Update image_apis.md to add quick tip for capturing physics state in sync with a call to an image API.

## How Has This Been Tested?

## Screenshots (if appropriate):
closed 2021-03-23 by jonyMarino