microsoft/AirSim #1498: Lidar initialization: pitch, roll, yaw

fernandodamasio, 2018-10-26:
Hi,

I'm coding an algorithm to build a HD Map of my environment.

I'm using `simGetGroundTruthKinematics `to return position and orientation (pitch, roll, yaw).

In the lidar.md file, it's written:

> - The API returns a Point-Cloud as a flat array of floats along with a timestamp of the capture.
> - The floats represent [x,y,z] coordinate for each point hit within the range in the last scan.
> - The coordinates are in the local vehicle NED like all other AirSim APIs.

My question is:

If I initialize the Lidar in settings.json defining x, y, z, pitch, roll, yaw with values, different then default values, the point cloud will be defined taking in consideration the rotation and position of the sensor or will return the coordinates in the local vehicle NED anyway?

Best regards,
Fernando Damasio

bmalhigh, 2018-10-29:
Fernando, right now, the Point-Cloud that is returned has points in the vehicle frame.
Based on your use case, is that reasonable?

fernandodamasio, 2018-10-29:
Hi @bmalhigh 

Right now it's reasonable to work with the lidar in the vehicle frame. TKS!

However, I have more questions about it.

I just wrote a code to create an HD Map of my environment. Next, you can find links to help understand the problem:

- Here you can see a video of the environment: https://www.youtube.com/watch?v=oDBrac6QwOs
- Lidar mapping code: https://github.com/skoods-org/welcome/blob/master/PythonExamples/lidar_mapping.py
- The project: https://github.com/skoods-org/welcome

1. I can see that the lidar returns the point cloud also taking into consideration the offset and rotation of the car. What I mean is, when driving the car and getting the points, the points will be updated based on the initial position of the car. Is this true? Because in real life, the points are always relative to the sensor position and we have to calculate the offset and rotation using other sensors, like the IMU.

2. What are the default values of the Lidar settings for the .json file?

3. I drew the track and guardrail of my environment using a mesh and a spline. But the lidar is not returning the points of this mesh. I know it because the guardrail is not in the point cloud. Do you know how to solve it? Maybe something related to collisions in Unreal? I'm including 2 images...

[image]

[image]

4. Do you have a helper code to rotate and translate the points to become relative to the sensors?

I appreciate your help.

Best regards,
Fernando Damasio

fernandodamasio, 2018-10-30:
Hi @bmalhigh 

2. TKS!

3. The spline mesh I'm using has a collision defined when drawing the spline, programmatically. It's free in the Unreal Launcher.

[image]

Do you see a way to work lidar with it? I don't have too much knowledge about creating meshes.

About 1. and 4. I will wait for your return.

TKS!!!

Regards,
Fernando Damasio

dahewujiang, 2018-11-02:
Hi, I want to sure that which coordinate system is adopted to represent the coordinate of  each point in the points cloud. Is the play start's coordinate system? I find that the car's position captured by the "record" button seems not in accord  with the position captured by lidar.

closed 2021-04-30 by jonyMarino