microsoft/AirSim #1499: Unreal crash after settings some parameters on settings.json

BrunoXL, 2018-10-26:
I tried to add a new camera throughout `settings.json`. I'm running on Ubuntu 16.04(xenial), AirSim 1.2.3 and UE 18.04. However, UE crash with this error:

  ` Exception was "SIGSEGV: invalid attempt to read memory at address 0x0000000000000178" `  

 Here is my log output: 

    [2018.10.26-18.55.06:728][235]LogLoad: Game class is 'AirSimGameMode'
	[2018.10.26-18.55.06:797][235]LogWorld: Bringing World /Game/UEDPIE_0_Rodain.Rodain up for play (max tick rate 0) at 2018.10.26-15.55.06
	[2018.10.26-18.55.06:797][235]LogWorld: Bringing up level for play took: 0.063295
	[2018.10.26-18.55.06:861][235]LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0009 seconds. Added 11 classes and 0 folders.
	[2018.10.26-18.55.06:866][235]Cmd: r.CustomDepth 3
	[2018.10.26-18.55.06:866][235]r.CustomDepth = "3"
	[2018.10.26-18.55.06:868][235]LogActor: Warning: BP_CameraDirector_C /Game/UEDPIE_0_Rodain.Rodain:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
	[2018.10.26-18.55.06:879][235]LogPawn: Error: EnableInput can only be specified on a Pawn for its Controller
	[2018.10.26-18.55.06:894][235]LogStats: FPlatformStackWalk::StackWalkAndDump -  0.014 s
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: === Handled ensure: ===
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: 
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: Ensure condition failed: !RotationQuat.ContainsNaN() [File:/home/bruno/repos/UnrealEngine/Engine/Source/Runtime/Core/Private/Math/UnrealMath.cpp] [Line: 429]
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: Invalid input to FRotator::Quaternion - generated NaN output: X=-nan Y=-nan Z=-nan W=nan
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: Stack: 
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b679a506f FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b6784ad35 FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b679a500c FLinuxPlatformStackWalk::StackWalkAndDumpEx(char*, unsigned long, int, unsigned int, void*)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b679ec9a2 FDebug::EnsureFailed(char const*, char const*, int, wchar_t const*)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b679ed53f FDebug::OptionallyLogFormattedEnsureMessageReturningFalse(bool, char const*, char const*, int, wchar_t const*, ...)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b679b865f FRotator::Quaternion() const
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e3054c PawnSimApi::createCamerasFromSettings()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e2fc91 PawnSimApi::setupCamerasFromSettings(common_utils::UniqueValueMap<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, APIPCamera*> const&)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e2f649 PawnSimApi::PawnSimApi(PawnSimApi::Params const&)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e4d5c5 ASimModeComputerVision::createVehicleSimApi(PawnSimApi::Params const&) const
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e41b3d ASimModeBase::setupVehiclesAndCamera()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e3e99b ASimModeBase::BeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b639beb4e AActor::DispatchBeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b639bdffc AActor::PostActorConstruction()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b639bbc04 AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b639b9a5c AActor::PostSpawnInitialize(FTransform const&, AActor*, APawn*, bool, bool, bool)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b642949fc UWorld::SpawnActor(UClass*, FTransform const*, FActorSpawnParameters const&)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b64295546 UWorld::SpawnActor(UClass*, FVector const*, FRotator const*, FActorSpawnParameters const&)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e3ab06 ASimHUD::createSimMode()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7ab5e3a042 ASimHUD::BeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b639beb4e AActor::DispatchBeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b64c842bd AWorldSettings::NotifyBeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b6412013c AGameStateBase::HandleBeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b64c6e749 UWorld::BeginPlay()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b640fe8cc UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5e15a19f UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5e152e30 UEditorEngine::PlayInEditor(UWorld*, bool)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5e146552 UEditorEngine::StartQueuedPlayMapRequest()
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5dac5b50 UEditorEngine::Tick(float, bool)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5e4ef7e6 UUnrealEdEngine::Tick(float, bool)
	[2018.10.26-18.55.06:894][235]LogOutputDevice: Error: [Callstack] 0x0000000000422859 FEngineLoop::Tick()
	[2018.10.26-18.55.06:895][235]LogOutputDevice: Error: [Callstack] 0x0000000000429fd3 GuardedMain(wchar_t const*)
	[2018.10.26-18.55.06:895][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5c398ea4 CommonLinuxMain(int, char**, int (*)(wchar_t const*))
	[2018.10.26-18.55.06:895][235]LogOutputDevice: Error: [Callstack] 0x00007f7b5b1d9830 /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f7b5b1d9830]
	[2018.10.26-18.55.06:895][235]LogOutputDevice: Error: [Callstack] 0x0000000000416b59 ./UE4Editor(_start+0x29) [0x416b59]
	[2018.10.26-18.55.06:895][235]LogOutputDevice: Error:  

That's my settings.json

         { 
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "SimMode": "ComputerVision",
    "ViewMode": "Manual",
    "Vehicles": {
      "SimpleFlight": {
        "VehicleType":"SimpleFlight",
        "DefaultVehicleName": "Armed",
        "AutoCreate": true,
        "PawnPaths": "",
        "EnableCollisionsPassthrough": false,
        "EnableCollisions": true,
        "AllowAPIAlways": true,
        "RC":{
          "RemoteControlID": 0,
          "AllowAPIWhenDisconnectd": false
        }
      },
      "Retrieve_Images": {
        "VehicleType":"ComputerVision",
        "DefaultVehicleName": "Armed",
        "AutoCreate": true,
        "PawnPaths": "",
        "EnableCollisionsPassthrough": false,
        "EnableCollisions": true,
        "AllowAPIAlways": true,
        "RC":{
          "RemoteControlID": 0,
          "AllowAPIWhenDisconnectd": false
        },
        "Cameras":{
          "MyCamera1":{
            "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 1920,
              "Height": 1080,
              "FOV_Degrees": 90,
              "AutoExposureSpeed": 100,
              "AutoExposureBias": 0,
              "AutoExposureMaxBrightness": 0.64,
              "AutoExposureMinBrightness": 0.03,
              "MotionBlurAmount": 0,
              "TargetGamma": 1.0,
              "ProjectionMode": "",
              "OrthoWidth": 5.12
            }
          ], 
          "X": 0, "Y": 0, "Z": 0,
          "Roll": 90, "Pitch": 0, "Yawl": 0 
          }
        }
      }  
    }
   }

msb336, 2018-12-18:
```
"Roll": 90, "Pitch": 0, "Yawl": 0 
```
Correct Yawl to Yaw and see if it works for you.
```
"Roll": 90, "Pitch": 0, "Yaw": 0

msb336, 2018-12-19:
@BrunoXL Can you post the snippet of your code related to image capture?

msb336, 2019-01-08:
@BrunoXL Are you still having problems related to this issue? If not, let's go ahead and close it.

jonyMarino, 2020-08-14:
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-14 by jonyMarino