microsoft/AirSim #1503: Invalid recorded images

MahirGulzar, 2018-10-28:
Hello devs, I recently came across this amazing package. I was playing around to extract camera image of type 0, 3, 5 . The problem I am facing is that while recording its writing invalid images i.e the windows is unable to recognize generated images. Its not happening consistently, some images do appear as they should while most of them are corrupted some how.

Setting.json is as follows:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "CaptureSettings":[ 
		{
			"ImageType": 0,
			"Width": 640,
			"Height": 480,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 100,
			"AutoExposureBias": 0,
			"AutoExposureMaxBrightness": 0.64,
			"AutoExposureMinBrightness": 0.03,
			"MotionBlurAmount": 0,
			"TargetGamma": 1,
			"ProjectionMode": "",
			"OrthoWidth": 5.12
		},
		{
			"ImageType": 5,
			"Width": 640,
			"Height": 480
		},
		{
			"ImageType": 3,
			"Width": 640,
			"Height": 480
		}
	],
  "Recording": {
    "Cameras": [
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": false },
		{ "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": false },
		{ "CameraName": "0", "ImageType": 3, "PixelsAsFloat": false, "Compress": false }
    ]
  }
}

any insights?

diarmaidocualain, 2018-10-31:
Hi! I am afraid I have not come across the same issue, but just to suggest a possible workaround; could you try slowing down the simulator? Maybe it is a timing issue. You can change the clockspeed in your settings file. 	"ClockSpeed": 0.5,

jonyMarino, 2020-08-14:
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

Zartris, 2020-08-17:
I'm having the same issue here and slowing down the clock speed does not fix it.
It is producing some images that is not corrupted, but in my case half is.
I tried recording images every 3rd second, but around half is still corrupted.
So the time to process this is not the problem.

My settings is almost the same, but I will paste them as well.
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 0.5,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 3,
    "Cameras": [
      {
        "CameraName": "center_custom",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "center_custom",
        "ImageType": 5,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 0,
              "ImageType": 0,
              "Width": 128,
              "Height": 72,
              "FOV_Degrees": 90,
              "AutoExposureSpeed": 100,
              "AutoExposureBias": 0,
              "AutoExposureMaxBrightness": 1,
              "AutoExposureMinBrightness": 1,
              "MotionBlurAmount": 0,
              "TargetGamma": 1.5
            },
            {
              "ImageType": 5,
              "Width": 128,
              "Height": 72,
              "FOV_Degrees": 90,
              "AutoExposureSpeed": 100,
              "AutoExposureBias": 0,
              "AutoExposureMaxBrightness": 1,
              "AutoExposureMinBrightness": 1,
              "MotionBlurAmount": 0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.5,
          "Y": 0.0,
          "Z": 0.0,
          "Pitch": 0.0,
          "Roll": 0.0,
          "Yaw": 0.0
        }
      }
    }
  },
  "SubWindows": [
    {
      "WindowID": 1,
      "ImageType": 5,
      "CameraName": "center_custom",
      "Visible": false
    },
    {
      "WindowID": 2,
      "ImageType": 0,
      "CameraName": "center_custom",
      "Visible": false
    }
  ]
}
```
Here we can see all the images that is corrupt (The once where image is not shown).
[image]