microsoft/AirSim #1506: Change car model

baaaze, 2018-10-30:
If I want to change my car model to another type. I have a mesh, physics, animations etc. What should I keep in mind? I've tried just switching the mesh but that doesn't work.

RVBldr, 2018-11-05:
There are several different files that are needed be linked together including not only the mesh, but there's a master blueprint, physics, animation, and tire data. Unreal has a series of videos for car creation that are really helpful: https://www.youtube.com/playlist?list=PLZlv_N0_O1gaz6eETtIxcmUDnqu5A-8At.  Take a look at those for a start.  There's another series by a guy named MathB that are also informative. If you're creating your own environment, see this page on Pawn Paths: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#pawnpaths.  Hope this helps a little.

RVBldr, 2018-12-07:
@baaaze did you have a chance to take a look at this?

GuiBo79, 2019-05-06:
Hi, when I try to load a custom car from Unreal Store I receive this message from Unreal when I try to run the simulator.

"There were no compatible vehicles created for current SimMode! Check your settings.json."

Below is my settings.json

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "DefaultCar": {"PawnBP":"Class'/Game/SportsCar/BP_SportsCar.BP_SportsCar_C'"},
  "EngineSound": true,
  "Vehicles": {
    "SkoodsI": {
      "VehicleType": "PhysXCar",
      "EnableCollisionPassthrough": true,
      "X": 0.0,
      "Y": 0.0,
      "Z": 0.0
    }
  },
  "DefaultSensors": {
    "Lidar": {
      "SensorType": 6,
      "Enabled": false,
      "DrawDebugPoints": false
    }
  }
}

What is a compatible vehicle ?

msb336, 2019-05-06:
In order to use different vehicle models, you have to specify the pawn path to the vehicle blueprint. You can see how that is done [here](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#available-settings-and-their-defaults). For your settings, I would probably have something like the following:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "PawnPaths" : {
    ""PawnBP":{"Class'/Game/SportsCar/BP_SportsCar.BP_SportsCar_C'"}
  },
  "Vehicles" : {
    "SkoodsI": {
        "VehicleType": "PhysXCar",
        "PawnPath" : "PawnBP"
        }
  }
}
```

GuiBo79, 2019-05-11:
> @baaaze did you have a chance to take a look at this?

Yes, I could Pawn my custom car but still having some issues. I can communicate with AirSim, get car state, control the car with the keyboard but unfortunately AIrSim API is not controlling the car. I just change for AirSim default car , everything works just fine again.

msb336, 2019-05-15:
Are you using the vehicle name in your API calls, your just calling with default vehicle input (`vehicle_name=''`)? If there is only one vehicle in your scene it shouldn't matter, but if there are more you will need to reference the specific vehicle you want to control.

thejeshk, 2020-02-23:
Hi Guys, 

I have my own vehicle model with the following skeleton [see image]. I have created a child blueprint class from AIRSIM carpawn and selected a mesh of my own vehicle. What else I should do in addition to that to use my vehicle ? 

[image]

RVBldr, 2020-02-23:
It’s not clear if you’ve been able to set up and drive the car outside of AirSim, or not, but there are a variety of tutorials on how to set up vehicles. See some of these:

https://www.youtube.com/watch?time_continue=9&v=A79_RN329xA&feature=emb_logo
https://www.youtube.com/watch?v=zGHGSwzhUr4&t=164s
                                                                                                                                               .
Once you can drive a vehicle, tuning can be a very iterative process and take some time. When you have the entire file structure of the vehicle in your project directory, you’ll want to use PawnPath to use that vehicle with AirSim.  https://microsoft.github.io/AirSim/docs/settings/#pawnpaths

[my_own_vehicle_skeleton]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F33362523%2F75112358-62e20a80-5643-11ea-99f5-e5babbb8db5d.PNG&data=02%7C01%7Cjim.piavis%40microsoft.com%7Cacb8c1bcd7344e48dd2f08d7b85eec4e%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637180590157477013&sdata=SaoXecy%2FzGvzYW5RzjgZ%2FC6RCyFA5k%2FAojI5o8CYFAI%3D&reserved=0>

thejeshk, 2020-02-24:
Hi @RVBldr 

Thank you for the fast response. Driving the vehicle is no problem. I have Created a new Vehicle project -> Imported Fbx mesh with Physics -> Used Vehicle car Blue print -> Created Animation BP accordingly -> Mapped Keyboard inputs and I could able to drive my car in the Level

BuT I could not able to bring my Car into AIRSIM :( 
I tried all the above steps. 

My Vehicle has slightly different bone names comparatively. Is it because of that or I have my own project. Should I do any modifications with my Project. 

I am not sure what I am supposed to do.

jvr-nava, 2020-03-24:
Is there a way of implementing a custom vehicle different from the PhysX car model. I have a couple of multi-wheeled and tracked actor blueprints with heavily customized physics and animations. I'd like to interface with AirSim's sensors. Any pointers would be much appreciated.

RVBldr, 2020-03-25:
For some of those, I would search for some of the Unreal Youtube videos. I do know there’s several on tracked vehicles, however, you may need to set those up to use the PhysX physics, but creating your own model.

jvr-nava, 2020-03-25:
@RVBldr wow, thanks for the fast answer. I think I didn't explain myself, I do have a working vehicle which uses customs c++ physics libraries to define my own track friction forces, as well as the vehicle dynamics and track animations. These are a combination of unreal blueprints and C++ code, and work reasonably well stand-alone. My question is more geared towards knowing if it is possible to integrate this 'custom' tracked vehicle with AirSim. From what I was able to find in this thread and other AirSim threads it seems that it is only possible to replace the standard AirSim car model with other model inheriting the built-in 'VehicleMovement' class (which seems to be a modified 'WheeledVehicleMovementComponent' that I'm not using since it does not provide the capabilites I needed for a tracked vehicle). But if it's possible are there any resources or pointers to where to start? 
Thanks!

RVBldr, 2020-03-25:
Airsim should work against a car class, however, instead of just replacing the existing SUV, have you tried the PawnPaths? https://microsoft.github.io/AirSim/docs/settings/#pawnpaths We’ve used this method for adding a separate car on Coastline as well as dealing with various forklifts in Warehouse, but they were all based on the same car class.

shaoxiang, 2020-05-29:
Hello @RVBldr 
I'm doing forklift simulation, can you share how you do it? Thank you very much.

tlgkyn06, 2020-05-30:
Hello @RVBldr and everyone
I suppose I created vehicle perfectly (-> Imported Fbx mesh with Physics -> Used Vehicle car Blue print -> Created Animation BP accordingly)  but when I try to play I face with "Error at startup: invalid map<K, T> key" error message . Where did I do wrong? Anybody to help?